thingsboard/thingsboard-gateway #355: [BUG]mqtt.json get data

Bruce1221, 2020-07-31:
**Describe the bug**
When i upgrade the gateway and restart it.The data was error.Look like the gateway parse the "key" too.
[image]

**Connector name:**
mqtt data parse error
**Details**
```
The device send json data like 
{"DeviceType":"Temp","DeviceID":"Temp_Server_1","TimeStamp":"2020-07-31 15:49:00","Temperature":27.90,"Humidity":38.40}

The mqtt.json is

 {
    "broker": {
      "name":"MQTT",
      "host":"192.168.0.1",
      "port":1883,
      "security": {
        "type": "basic",
        "username": "user",
        "password": "user"
      }
    },
    "mapping": [
        {
            "topicFilter": "Device/Temp/#",
            "converter": {
                "attributes": [
                    {
                        "type": "string",
                        "key": "DeviceID",
                        "value": "${DeviceID}"
                    }
                ],
                "timeseries": [
                    {
                        "type": "double",
                        "key": "Temperature",
                        "value": "${Temperature}"
                    },
                    {
                        "type": "double",
                        "key": "Humidity",
                        "value": "${Humidity}"
                    }
                ],
                "type": "json",
                "deviceNameJsonExpression": "${DeviceID}",
                "deviceTypeJsonExpression": "${DeviceType}"
            }
        }
    ],
    "connectRequests": [
      {
        "topicFilter": "device/connect",
        "deviceNameJsonExpression": "${DeviceID}"
      },
      {
        "topicFilter": "device/+/connect",
        "deviceNameTopicExpression": "(?<=device\/)(.*?)(?=\/connect)"
      }
    ],
    "disconnectRequests": [
      {
        "topicFilter": "device/disconnect",
        "deviceNameJsonExpression": "${DeviceID}"
      },
      {
        "topicFilter": "device/+/disconnect",
        "deviceNameTopicExpression": "(?<=device\/)(.*?)(?=\/disconnect)"
      }
    ],
    "attributeUpdates": [
      {
        "deviceNameFilter": "SmartMeter.*",
        "attributeFilter": "uploadFrequency",
        "topicExpression": "device/${DeviceID}/${attributeKey}",
        "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
      }
    ],
    "serverSideRpc": [
      {
        "deviceNameFilter": ".*",
        "methodFilter": "echo",
        "requestTopicExpression": "device/${DeviceID}/request/${method}/${requestId}",
        "responseTopicExpression": "device/${DeviceID}/response/${method}/${requestId}",
        "responseTimeout": 10000,
        "valueExpression": "${params}"
      },
      {
        "deviceNameFilter": ".*",
        "methodFilter": "no-reply",
        "requestTopicExpression": "device/${DeviceID}/request/${method}/${requestId}",
        "valueExpression": "${params}"
      }
    ]
  }
```

**Versions (please complete the following information):**
 - OS: centos7
 - Thingsboard IoT Gateway version [2.0]
 - Python version[3.7]

sobehard, 2020-07-31:
Hi, there is one solution that can temporary solved your issue before this bug is resolved.
[image]
You just need to make sure that the param for key and value field is not the same. 
For example:-
"key": temperature, "value": "${Temperature}" or "key": Temperature, "value": "${temperature}"

Bruce1221, 2020-07-31:
I did that before issue.
But,the database already storage over 20 millions data.
So,i rollback the gateway to version 2.1.
Thank you.

zbeacon, 2020-07-31:
Hi @Bruce1221 

Thank you for your feedback.
Could you try to use the version from the master branch? Please try and let me know about the results.

Bruce1221, 2020-08-01:
Hi,
I modified the Version2.5 Gateway's  source code directly,It,s works well.
Thank you.

closed 2020-08-01 by Bruce1221