thingsboard/thingsboard-gateway #357: [BUG] Python error

MrPivato, 2020-07-31:
Error in python script from Thingsboard gateway

**Connector name:**
MQTT Connector

**Error traceback:**
```
""2020-07-31 14:24:24" - INFO - [tb_gateway_service.py] - tb_gateway_service - 72 - Gateway starting..."
""2020-07-31 14:24:24" - INFO - [tb_gateway_service.py] - tb_gateway_service - 77 - ThingsBoard IoT gateway version: 2.5.0"
""2020-07-31 14:24:24" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 1"
""2020-07-31 14:24:25" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 285 - Error on loading connector:"
""2020-07-31 14:24:25" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 286 - Expecting value: line 60 column 9 (char 1402)"
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-2.5.0-py3.8.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 279, in _load_connectors
    connector_conf = load(conf_file)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 457, in load
    return loads(fp.read(),
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 60 column 9 (char 1402)
""2020-07-31 14:24:25" - INFO - [tb_gateway_service.py] - tb_gateway_service - 141 - Gateway started."
""2020-07-31 14:24:25" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-07-31 14:24:25" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 2"
```

***tb-gateway.yaml***
```
thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: my_gateway_id_here
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
```

***mqtt.json***
```
{
  "broker": {
    "name":"Default Local Broker",
    "host":"139.200.55.111",
    "port":1883,
    "security": {
      "type": "basic",
      "username": "my_mqtt_broker_username_here",
      "password": "my_mqtt_broker_password_here"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Lamp",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          
        ],
        "timeseries": [
          {
            "type": "boolean",
            "key": "is_on",
            "value": "${is_on}"
          },
          {
            "type": "long",
            "key": "lamp_off_value",
            "value": "${lamp_off_value}"
          },
          {
            "type": "long",
            "key": "lamp_on_value",
            "value": "${lamp_on_value}"
          },
          {
            "type": "long",
            "key": "lux_treshold",
            "value": "${lux_treshold}"
          },
          {
            "type": "long",
            "key": "movement_threshold",
            "value": "${movement_threshold}"
          },
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
```

***MQTT message***
```
{
   "topic":"/sensor/c3d09eb4-c5f7-11ea-87d0-0242ac130003/data",
   "payload":"{\"sensorModel\":\"ESP8266\",\"is_on\":true,\"lamp_off_value\":0,\"lamp_on_value\":1000,\"lux_treshold\":80,\"movement_threshold\":10}",
   "qos":0,
   "retain":false,
   "_msgid":"50b49c51.37ba84"
}
```
 - OS: Ubuntu 20.04.1 LTS x86_64
 - Thingsboard IoT Gateway version:  master branch (commit 7d30d455e0c9c1a9bf832c9ae9d513d340aedc20)
 - Python version 3.8.2

zbeacon, 2020-08-01:
Hi @MrPivato,

Thank you for your interest in the ThingsBoard IoT gateway.

This error caused by wrong format in the mqtt.json to check your json, you can use any online json formatter, it will show to you errors in the json.

MrPivato, 2020-08-01:
Hi!
In the`mqtt.json` or in the json I am sending?

zbeacon, 2020-08-03:
Hi @MrPivato 

In the json you sent. But anyway the gateway throws the following error: `JSONDecodeError` - this error means the gateway cannot parse the json configuration file.

MrPivato, 2020-08-03:
Thanks for the insight!
I will try to fix it now knowing that. 
Sorry for labeling it as a bug, cleary my fault 😅.

zbeacon, 2020-08-04:
Hi @MrPivato 

I close the issue, if you will have any questions about the gateway, please open the new one.

closed 2020-08-04 by zbeacon

frank-wettstein, 2024-01-17:
"We have the same problem which leads to the thread dying. However, this should not be the case. The thread should simply discard the message, handle the exception, and log it."