thingsboard/thingsboard-gateway #358: [HELP]

krishna194, 2020-08-03:
**Describe the issue**
Create description about your issue, and your actions to solve it.
Unable to connect thingsboard gateway using REST connector
I have python Rest test server on laptap
I want to connect it with thingsboard through gateway and acess data using POST,GET,PUT methods

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. REST Connector]

**Error traceback (If it was raised):**
```no data is received 

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]
[rest.txt](https://github.com/thingsboard/thingsboard-gateway/files/5016417/rest.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/5016419/tb_gateway.txt)

zbeacon, 2020-08-04:
Hi @krishna194 

Thank you for your interest in the ThingsBoard IoT gateway.

REST connector gives ability to create endpoints on the gateway and listen ports to get information.
To take the information from another resource you should use [Request connector](https://thingsboard.io/docs/iot-gateway/config/request/).

krishna194, 2020-08-04:
Thanks for the quick response
I have a sample rest endpoint server running on my pc with endpoint
127.0.0.1:5000/user/username
How to get information with request connector
My request configuration is attached below

I have another doubt
Whether REST and Request connectors can be enabled at the same time through
gateway.?

{
  "host": "http://127.0.0.1:5000",
  "SSLVerify": true,
  "security": {
    "type": "basic",
    "username": "user",
    "password": "password"
  },
  "mapping": [
    {
      "url": "/user/username",
      "httpMethod": "GET",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 0.5,
      "scanPeriod": 5,
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "SD8500",
        "deviceTypeJsonExpression": "SD",
        "attributes": [
          {
            "key": "serialNumber",
            "type": "string",
            "value": "${serial}"
          }
        ],
        "telemetry": [
          {
            "key": "Maintainer",
            "type": "string",
            "value": "${Developer}"
          }
        ]
      }
    },
    {
      "url": "/user/username",
      "httpMethod": "GET",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 0.5,
      "scanPeriod": 100,
      "converter": {
        "type": "custom",
        "deviceNameJsonExpression": "SD8500",
        "deviceTypeJsonExpression": "SD",
        "extension": "CustomRequestUplinkConverter",
        "extension-config": [
          {
            "key": "Totaliser",
            "type": "float",
            "fromByte": 0,
            "toByte": 4,
            "byteorder": "big",
            "signed": true,
            "multiplier": 1
          },
          {
            "key": "Flow",
            "type": "int",
            "fromByte": 4,
            "toByte": 6,
            "byteorder": "big",
            "signed": true,
            "multiplier": 0.01
          },
          {
            "key": "Temperature",
            "type": "int",
            "fromByte": 8,
            "toByte": 10,
            "byteorder": "big",
            "signed": true,
            "multiplier": 0.01
          },
          {
            "key": "Pressure",
            "type": "int",
            "fromByte": 12,
            "toByte": 14,
            "byteorder": "big",
            "signed": true,
            "multiplier": 0.01
          },
          {
            "key": "deviceStatus",
            "type": "int",
            "byteAddress": 15,
            "fromBit": 4,
            "toBit": 8,
            "byteorder": "big",
            "signed": false
          },
          {
            "key": "OUT2",
            "type": "int",
            "byteAddress": 15,
            "fromBit": 1,
            "toBit": 2,
            "byteorder": "big"
          },
          {
            "key": "OUT1",
            "type": "int",
            "byteAddress": 15,
            "fromBit": 0,
            "toBit": 1,
            "byteorder": "big"
          }
        ]
      }
    }
  ],
  "attributeUpdates": [
      {
        "httpMethod": "POST",
        "httpHeaders": {
          "CONTENT-TYPE": "application/json"
        },
        "timeout": 0.5,
        "tries": 3,
        "allowRedirects": true,
        "deviceNameFilter": "SD.*",
        "attributeFilter": "send_data",
        "requestUrlExpression": "sensor/${deviceName}/${attributeKey}",
        "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
      }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "httpMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    }
  ]
}

zbeacon, 2020-08-04:
>> How to get information with request connector
To get information from some endpoint, you should configure the converter section. It configuration depends on response from server (json/bytes). To get information from it you can use just ${attribute_name} to get attribute_name parameter value from the incoming message or use JSONPathExpression e.g. in case if you get the array of values. 

>> Whether REST and Request connectors can be enabled at the same time through
gateway.?

Yes, you can do it. Also you can run several connectors with the same type (e.g. 2 request connectors and one rest connector). To do this uncomment/copy connector section in the tb_gateway.yaml file. Only one requirement - every connector must has his own configuration file.

krishna194, 2020-08-04:
Thanks for your reply
I have another question regarding the ble connector
I have a ble simulator on smartphone,parametes like heart rate ,temperature
etc
I want to get the UUID of these parameters using python script,but it shows
connection refused error
Can you explain the reason or another method for this?

zbeacon, 2020-08-05:
Hi @krishna194 ,

Please open a new issue, when you wanna ask something new. 

Probably, the issue with simulator is appeared because the smartphone usually doesn't support BLE connection to them without pairing, just try to pair device with gateway and your smartphone and try to run the gateway again.

closed 2020-08-05 by zbeacon