thingsboard/thingsboard-gateway #360: [HELP]

Dhanya1995, 2020-08-04:
REST CONNECTOR ERROR

I have created a rest server on kepware on my PC. I tried to configure the rest.json and I am getting error. I have uploaded the details here. 

 - OS: [Windows 10]
 - Thingsboard IoT Gateway version 2.5.0
 - Python version 3.8.3

[restconfig.txt](https://github.com/thingsboard/thingsboard-gateway/files/5020056/restconfig.txt)

[error.txt](https://github.com/thingsboard/thingsboard-gateway/files/5020057/error.txt)

zbeacon, 2020-08-04:
Hi @Dhanya1995 

Thank you for your interesnt in the ThingsBoard IoT gateway.

REST connector gives ability to create endpoints on the gateway and listen ports to get information.
To take the information from another resource you should use [Request connector](https://thingsboard.io/docs/iot-gateway/config/request/).

Dhanya1995, 2020-08-04:
Thank you for the quick reply.

closed 2020-08-04 by Dhanya1995