thingsboard/thingsboard-gateway #362: [BUG] - MQTT connector - duplicit telemetries

zdenoL, 2020-08-05:
**from time to time happens, that in some moment, each telemetry from MQTT connected device is sent twice**
Clear and concise description of what the bug is.

**MQTT connector**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'2020-08-05 06:59:31' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 06:59:31' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:09:30' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:09:30' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:19:29' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:19:29' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:29:31' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:29:32' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:39:28' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:39:28' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:49:29' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'
''2020-08-05 07:49:30' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic TEMPMON/TEMP001/telemetry'

```

**Versions:**
 - OS:20.04
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version  3.7

zbeacon, 2020-08-06:
Hi @zdenoL ,

Thank you for your interest in the Thingsboard IoT gateway. 
At first point please update your gateway to the lates version. If it won't help, please change the logging level from INFO to DEBUG in the logs.conf file, repeat the issue and send logs.

zdenoL, 2020-08-08:
Hi, do you have any docs guide how to update dockerized dgateway deployed with docker compose?

Nadou14, 2020-08-11:
@zbeacon  I'm also facing a problem with mapping, I'm using the default configuration and everything wend fine, but when I've tried to add some key values in the config file they weren't parsed in my telemetry device, here's my mapping config file:

> "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          },
        {
            "type": "double",
            "key": "longitude",
            "value": "${$.long}"
          }
        ]
      }
    },
and this is the body that I've been trying to parse:
`mosquitto_pub -d -h 127.0.0.1 -p 1884 -t "/sensor/data" -m '{"serialNumber":"1235", "sensorType":"Thermometer", "sensorModel":"ciao", "temp":12,"hum":33, "long":44.555}'`
[image]

only the values of temperature and humidity were parsed but not the value of longitude, I replaced long with longitude, didn't work either. I've tried many other values and keys but still stuck in the same problem, any suggestions!

zbeacon, 2020-09-15:
Hi @zdenoL ,

I have added example of the docker-compose.yaml file into docker folder in the repository, you can use it to build docker-compose, you need.

closed 2020-09-15 by zbeacon