thingsboard/thingsboard-gateway #364: [HELP] POST not working with REST connector

Dhanya1995, 2020-08-05:
I had created an endpoint using rest connector. I want to post some data from thingsboard to this endpoint. I have a device and it has values in it . The endpoint displays the following message:

{"message": "The server does not support the media type transmitted in the request."}

The tb_gateway shows :

 A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.

This is my configuration:
{

      "endpoint": "/device_trial",
      "HTTPMethod": [
        "POST"
      ],
      "security":
      {
        "type": "anonymous",
        
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "Device ${Device 1}",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    }
    
  ]
}

 - OS: [Windows10]
 - Thingsboard IoT Gateway version [e.g. 2.5]
 - Python version[e.g. 3.8]

Could you please give me any suggestions

zbeacon, 2020-08-06:
Hi @Dhanya1995 ,

Thank you for your interest in the ThingsBoard IoT gateway.

Did you add the CONTENT-TYPE header into your request? Please try to add the CONTENT-TYPE header into your request with value "application/json".

Dhanya1995, 2020-08-06:
Thankyou for the reply. 

I have added content type. Still getting error.

I have started learning thingsboard only recently and I want to understand how this rest connector works.  I donot have any physical device connected. I have some devices on my thingsboard, which has certain attributes. Is it possible to bring these values to the created endpoint by using POST. 

I am always getting this error : A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

on the endpoint I have created , I am getting {"message": "The server does not support the media type transmitted in the request."}.  I have attached my config file as well. 

[rest configuration.txt](https://github.com/thingsboard/thingsboard-gateway/files/5033655/rest.configuration.txt)

zbeacon, 2020-08-07:
Hi @Dhanya1995 ,

Please attach also request you sent and logs from the connector.log file. Probably some exceptions are there.

Dhanya1995, 2020-08-07:
My connector log file is not showing anything recently. It is not displaying anything since the past few days. so is the connector not working?? But, an point is created. I have attached my tb_gateway output here.

Is there any sample configuration files that I could use? 

[query.txt](https://github.com/thingsboard/thingsboard-gateway/files/5039597/query.txt)
[request.txt](https://github.com/thingsboard/thingsboard-gateway/files/5039598/request.txt)

zbeacon, 2020-08-07:
It looks like you looking for logs in the wrong location.

I mean not a attributeRequests section from the configuration, I mean request, you sent to the endpoint (headers, body, authorization, etc.)

Dhanya1995, 2020-08-07:
Thankyou for the response.

I used postman to sent request. I am attaching the authorisation, header and body. The postman output shows "OK". But it is not visible at the endpoint(localhost:5000/mydevice).

[image]
[image]
[image]

this is the message received at the endpoint:
{"message": "The server does not support the media type transmitted in the request."}

zbeacon, 2020-08-07:
@Dhanya1995  

Your message doesn't contain "name" field, but it is in the configuration. Just add this field to the request or remove ${name} from deviceNameJsonExpression value in the configuration.

Dhanya1995, 2020-08-07:
Thankyou.

I changed the format like this. Still getting the error. I have added my log file as well.
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/5040997/connector.log)
[image]

zbeacon, 2020-08-07:
The correct request for your configuration looks like the following:

```
{
  "name": "abc",
  "sensorModel": "YGBI212",
  "temp": 25,
  "hum": 78.6
}

```

Dhanya1995, 2020-08-07:
i am getting it. THankyou very much for the help

closed 2020-08-11 by Dhanya1995

Nadou14, 2020-08-12:
I'm facing the same issue even though I tried the same configuration but I still get the same server Error(500), I used the same message as suggested @zbeacon  and the default configuration in rest.json . I also used postman to post the data with basic authorzation. here's my config file rest.json 
[image]
and here's the body of my message:
[image]

Nadou14, 2020-08-13:
@Dhanya1995 how did you fix your problem because I'm facing the same issue?

Dhanya1995, 2020-08-13:
i didnt get server error. mine showed "server doesnot support the media type". Did you run the tb-gateway the python file before sending data through postman. I mean, it should show like "running on 127.0.0.1". Keep it running in the background and then post the data. Try doing that. I dont really know much about this.

Nadou14, 2020-08-13:
@Dhanya1995 Thank you for your reply, it was a configuration problem. It's working now.