thingsboard/thingsboard-gateway #368: POST method using request connector[HELP]

meghanajose, 2020-08-11:
Trying to work on POST method with Request connector. Not receiving data in thingsboard. I used Postman to post the data.
this is the body:
{
  "key":56,
  "id" :5
}

**Configuration** (thingsboard:
  host: demo.thingsboard.io
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: EAwVlViQR9c257A6R6CB
storage:
  type: memory
  read_records_count: 10
  max_records_count: 1000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json

#  -
#     name: Modbus Connector
#     type: modbus
#     configuration: modbus.json
#
#  -
#     name: Modbus Connector
#     type: modbus
#     configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
      name: REQUEST Connector
      type: request
      configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#   name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):** Request connector 
{
  "host": "https://reqres.in",
  "SSLVerify": true,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "url": "api/users",
      "httpMethod": "POST",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 0.5,
      "scanPeriod": 1,
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "test",
        "deviceTypeJsonExpression": "default",
        "attributes": [
          {
            "key": "serialNumber",
            "type": "string",
            "value": "${Id}"
          }
        ],
        "telemetry": [
          {
            "key": "Maintainer",
            "type": "string",
            "value": "${key}"
          }

        ]
      }
    }
  ],
  "attributeUpdates": [
    {
      "httpMethod": "POST",
      "httpHeaders": {
        "CONTENT-TYPE": "application/json"
      },
      "timeout": 0.5,
      "tries": 3,
      "allowRedirects": true,
      "deviceNameFilter": "SD.*",
      "attributeFilter": "send_data",
      "requestUrlExpression": "test/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "httpMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    }
  ]
}

    

**Error traceback (If it was raised):**
```
""2020-08-11 13:19:44" - ERROR - [json_request_uplink_converter.py] - json_request_uplink_converter - 52 - Cannot find the value from the input data"
""2020-08-11 13:19:44" - ERROR - [json_request_uplink_converter.py] - json_request_uplink_converter - 52 - Cannot find the value from the input data"
""2020-08-11 13:19:44" - ERROR - [tb_utility.py] - tb_utility - 62 - No telemetry and attributes in data: {"deviceName": "test", "deviceType": "default", "attributes": [], "telemetry": []}"
""2020-08-11 13:19:44" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 314 - Data from REQUEST Connector connector is invalid."

```

**Versions (please complete the following information):**
 - OS: [windows 10]
 - Thingsboard IoT Gateway version [2.5.0]
 - Python version[.3.7]

zbeacon, 2020-08-11:
Hi @meghanajose ,

Thank you for your interest in the ThingsBoard IoT gateway.
Request connector is developed to get information from the external REST API endpoints. In your case, you should use [REST connector](https://thingsboard.io/docs/iot-gateway/config/rest/).

meghanajose, 2020-08-11:
Hi @zbeacon , 
Thank you for your immediate response. So basically in this request connector only GET method is allowed?

zbeacon, 2020-08-12:
Hi @meghanajose ,

No the Request connector you can use to get some data from external API endpoints, if you want to push data to the gateway, you should use REST connector, it will create endpoints on the gateway and you will be able to push data to them. Both connectors support GET and POST requests, if some of them doesn't work it is a bug.

meghanajose, 2020-08-12:
Hi @zbeacon ,
Thanks for the response. Would you please share a configuration file of POST requests using REQUEST connector?

zbeacon, 2020-09-01:
Hi @meghanajose 

I apologyze for the long term answer, I have added the ability to send the data in the request, please try the version from the master branch and use the following configuration:

```
{
  "host": "https://reqres.in",
  "SSLVerify": true,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "url": "api/users",
      "httpMethod": "POST",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "content": {
        "name": "morpheus",
        "job": "leader"
      },
      "allowRedirects": true,
      "timeout": 0.5,
      "scanPeriod": 1,
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "test",
        "deviceTypeJsonExpression": "default",
        "attributes": [
          {
            "key": "serialNumber",
            "type": "string",
            "value": "${id}"
          }
        ],
        "telemetry": [
          {
            "key": "Maintainer",
            "type": "string",
            "value": "${name}"
          }

        ]
      }
    }
  ],
  "attributeUpdates": [
    {
      "httpMethod": "POST",
      "httpHeaders": {
        "CONTENT-TYPE": "application/json"
      },
      "timeout": 0.5,
      "tries": 3,
      "allowRedirects": true,
      "deviceNameFilter": "SD.*",
      "attributeFilter": "send_data",
      "requestUrlExpression": "test/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "httpMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    }
  ]
}

```

closed 2020-09-09 by zbeacon