thingsboard/thingsboard-gateway #369: [BUG] python3-pyrsistent is not installable, error in config

pehedima, 2020-08-11:
Hi together,

I tried to deploy the latest python3-thingsboard-gateway.deb package (2.5.0) and got the following error:

```
...
[Build]   main  ---> Running in e17944799242
[Build]   main --2020-08-11 10:13:53--  https://github.com/thingsboard/thingsboard-gateway/releases/latest/download/python3-thingsboard-gateway.deb
[Build]   main Resolving github.com (github.com)... 140.82.118.3
[Build]   main Connecting to github.com (github.com)|140.82.118.3|:443... connected.
[Build]   main HTTP request sent, awaiting response... 302 Found
[Build]   main Location: https://github.com/thingsboard/thingsboard-gateway/releases/download/2.5.0/python3-thingsboard-gateway.deb [following]
[Build]   main --2020-08-11 10:13:53--  https://github.com/thingsboard/thingsboard-gateway/releases/download/2.5.0/python3-thingsboard-gateway.deb
[Build]   main Reusing existing connection to github.com:443.
[Build]   main HTTP request sent, awaiting response... 302 Found
[Build]   main Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/78083065/d8cc8f80-ca84-11ea-9a18-4974b2f5cc34?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200811%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200811T081353Z&X-Amz-Expires=300&X-Amz-Signature=2a629cba37fbd9b28b4e347692e5c7ff5a8cb254627b4ada2d88bcfad3b62c95&X-Amz-SignedHeaders=host&actor_id=0&repo_id=78083065&response-content-disposition=attachment%3B%20filename%3Dpython3-thingsboard-gateway.deb&response-content-type=application%2Foctet-stream [following]
[Build]   main --2020-08-11 10:13:53--  https://github-production-release-asset-2e65be.s3.amazonaws.com/78083065/d8cc8f80-ca84-11ea-9a18-4974b2f5cc34?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200811%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200811T081353Z&X-Amz-Expires=300&X-Amz-Signature=2a629cba37fbd9b28b4e347692e5c7ff5a8cb254627b4ada2d88bcfad3b62c95&X-Amz-SignedHeaders=host&actor_id=0&repo_id=78083065&response-content-disposition=attachment%3B%20filename%3Dpython3-thingsboard-gateway.deb&response-content-type=application%2Foctet-stream
[Build]   main Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.146.115
[Build]   main Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.146.115|:443... connected.
[Build]   main HTTP request sent, awaiting response... 200 OK
[Build]   main Length: 75056 (73K) [application/octet-stream]
[Build]   main Saving to: ‘python3-thingsboard-gateway.deb’
[Build]   main      0K .......... .......... .......... .......... .......... 68%  267K 0s
[Build]   main     50K .......... .......... ...                             100% 4.32M=0.2s
[Build]   main 2020-08-11 10:13:54 (380 KB/s) - ‘python3-thingsboard-gateway.deb’ saved [75056/75056]
[Build]   main Removing intermediate container e17944799242
[Build]   main  ---> ecc06feb247f
[Build]   main Step 9/38 : RUN ["/tmp/qemu-execve","-execve","/bin/sh","-c","sudo apt-get install -y /python3-thingsboard-gateway.deb"]
[Build]   main  ---> Running in a7d713223224
[Build]   main Reading package lists...
[Build]   main Building dependency tree...
[Build]   main Reading state information...
[Build]   main Some packages could not be installed. This may mean that you have
[Build]   main requested an impossible situation or if you are using the unstable
[Build]   main distribution that some required packages have not yet been created
[Build]   main or been moved out of Incoming.
[Build]   main The following information may help to resolve the situation:
[Build]   main The following packages have unmet dependencies:
[Build]   main  python3-thingsboard-gateway : Depends: python3-pyrsistent but it is not installable
[Build]   main E: Unable to correct problems, you have held broken packages.
[Build]   Built 1 service in 3:17
[Error]   Deploy failed
...
```

I saw, that the error were already mentioned [here](https://github.com/thingsboard/thingsboard-gateway/issues/295). 
In the latest version the dependencies for the python3-pyrsistent package are inserted again. Why?

In this case it is not working. I deleted the dependency in the control file of the DEB package. Recreate the package and the installation was successful without any faults.

control:
```
Package: python3-thingsboard-gateway
Source: thingsboard-gateway
Version: 2.5.0-1
Architecture: all
Maintainer: ThingsBoard <info@thingsboard.io>
Installed-Size: 526
Depends: python3-jsonpath-rw, python3-paho-mqtt, python3-pip, python3-regex, python3-requests, python3-simplejson, python3-yaml, python3:any
Section: python
...
```

zbeacon, 2020-08-12:
Hi @pehedima ,

Thank you for your investigation I will regenerate the deb package and add it to the latest release.

zbeacon, 2020-08-12:
The latest release was updated, it contains new deb and rpm packages. Thank you for your participating.

closed 2020-08-12 by zbeacon