thingsboard/thingsboard-gateway #370: [HELP]

krishna194, 2020-08-11:
**Describe the issue**
Create description about your issue, and your actions to solve it.
I have to send data to things-board-gateway ble connector
I'm using BLE sim android application as ble simulator
while starting things-board-gateway,I'm unable to establish connection  
suggest a solution for this problem
**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
ble connector

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```
Ble working with only root user

**Versions (please complete the following information):**
 - OS: MInt 19.3
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]
[ble.txt](https://github.com/thingsboard/thingsboard-gateway/files/5056499/ble.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/5056500/tb_gateway.txt)
[connector.txt](https://github.com/thingsboard/thingsboard-gateway/files/5056501/connector.txt)
[service.txt](https://github.com/thingsboard/thingsboard-gateway/files/5056502/service.txt)

zbeacon, 2020-08-12:
Hi @krishna194 

Try to use the parameter with name deviceName instead of name in your BLE connector configuration file.

krishna194, 2020-08-12:
Thank you for your reply
I'm using a BLESIM simulator android app,
it is having three parameters Battery,Heart Monitor,Health thermometer
Can you suggest how to give the configuration accordingly.
I have to pair with my  bluettoth before starting thingsboard gateway.

krishna194, 2020-08-12:
ok
I have written my device name as same as in BLE configuration file
An error occurred while running gateway 
BLE work only on root command
I have given acess to all devices using the command
sudo setcap 'cap_net_raw,cap_net_admin+eip' /usr/local/lib/python3.6/dist-packages/bluepy/bluepy-helper
But the same error is repeating while restarting thingsboard-gateway
[ble.txt](https://github.com/thingsboard/thingsboard-gateway/files/5063110/ble.txt)
[connector.txt](https://github.com/thingsboard/thingsboard-gateway/files/5063111/connector.txt)

zbeacon, 2020-08-13:
Hi @krishna194 

Try please the following config:

```

{
    "name": "BLE Connector",
    "rescanIntervalSeconds": 100,
    "checkIntervalSeconds": 100,
    "scanTimeSeconds": 5,
    "passiveScanMode": true,
    "devices": [
        {
            "deviceName": "Temperature sensor",
            "MACAddress": "80:58:F8:35:D4:BC",
            "addrType": "public",
            "telemetry": [
                {
                    "key": "Battery",
                    "method": "notify",
                    "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
                    "byteFrom": 2,
                    "byteTo": 6
                }
            ],
            "attributes": [
                {
                    "key": "Battery",
                    "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
                    "method": "read",
                    "byteFrom": 0,
                    "byteTo": -1
                }
            ],
            "attributeUpdates": [
                {
                    "attributeOnThingsBoard": "sharedName",
                    "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB"
                }
            ],
            "serverSideRpc": [
                {
                    "methodRPC": "rpcMethod1",
                    "withResponse": true,
                    "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
                    "methodProcessing": "read"
                }
            ]
        }
    ]
}

```

krishna194, 2020-08-13:
I tried this configuration 
I have got the error" BLE work only on root command"
I tried this command-"sudo setcap 'cap_net_raw,cap_net_admin+eip' /usr/local/lib/python3.6/dist-packages/bluepy/bluepy-helper"
But the same error is repeating while restarting thingsboard-gateway
[connector.txt](https://github.com/thingsboard/thingsboard-gateway/files/5069439/connector.txt)
[ble(1).txt](https://github.com/thingsboard/thingsboard-gateway/files/5069447/ble.1.txt)

samson0v, 2021-12-09:
Hi @krishna194, we are remaking the BLE connector now. We will let you know when it is done.

Thanks for your interest in ThignsBoard IoT Gateway!

samson0v, 2021-12-17:
Hi @krishna194, we did a remake of the BLE Connector yesterday, so you can install the new version of Gateway from the master branch and test the connector.
Let us know if the new version of BLE Connector fixes your problem.

closed 2022-01-05 by zbeacon