thingsboard/thingsboard-gateway #371: [HELP] Unable to post data with Rest connector ( HTTP 500)

Nadou14, 2020-08-13:
I've configured Rest connector to connect to my thingsboard instance after trying mqtt connector, both connectors are enabled, but I'm having some issues while trying to post data to my endpoint I get 500 server error, even though I've checked the data format normally it is correct, 
in the log I get this error : in the line 343 in rect-connector.py in process data 
[image]

I was posting my data from postman 
[image]
and here's my rest.json file config
[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

zbeacon, 2020-08-13:
Hi @Nadou14 

Thank you for your interest in ThingsBoard IoT gateway. It looks like an issue with configuration, try to change HTTPMethod (parameter name) in your connector configuration file to HTTPMethods.

Nadou14, 2020-08-13:
@zbeacon Thank's a lot, I've changerd that and now it works.

closed 2020-08-13 by Nadou14