thingsboard/thingsboard-gateway #372: [HELP] BLE Connector - custom services not read

Extulen, 2020-08-13:
**Describe the issue**
Possible duplicate of #370.
When thingsboard-gateway runs, it connects to the device. Default services are found and the device is created on thingsboard with those attributes.
It then raises the errors below multiple times and stops communicating with the device.
```
Aug 13 14:42:03 rpi4cms python3[1328]: ""2020-08-13 14:42:03" - ERROR - [tb_utility.py] - tb_utility - 62 - deviceName is empty in data: {}"
Aug 13 14:42:03 rpi4cms python3[1328]: ""2020-08-13 14:42:03" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 314 - Data from BLE Connector connector is invalid."
```

This issue is happening with two different BLE devices (each with different custom services), both of which were confirmed working by using the nRFConnect app.

**Configuration**
```
{
    "name": "BLE Connector",
    "rescanIntervalSeconds": 20,
    "checkIntervalSeconds": 20,
    "scanTimeSeconds": 5,
    "passiveScanMode": true,
    "devices": [
        {
            "name": "BLE Device",
            "MACAddress": "CB:1B:3E:A8:BC:90",
            "deviceType": "BLEDevice",
            "telemetry": [
                {
                    "key": "Touch Count",
                    "method": "notify",
                    "characteristicUUID": "00000001-1212-efde-1523-785fef13d123",
                    "byteFrom": 0,
                    "byteTo": -1
                },
                {
                    "key": "Move Count",
                    "method": "notify",
                    "characteristicUUID": "00000002-1212-efde-1523-785fef13d123",
                    "byteFrom": 0,
                    "byteTo": -1
                }
            ],
            "attributes": [
                {
                    "key": "Device Name",
                    "method": "read",
                    "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb",
                    "byteFrom": 0,
                    "byteTo": -1
                },
                {
                    "key": "Appearance",
                    "method": "read",
                    "characteristicUUID": "00002a01-0000-1000-8000-00805f9b34fb",
                    "byteFrom": 0,
                    "byteTo": -1
                },
                {
                    "key": "Peripheral Preferred Connection Parameters",
                    "method": "read",
                    "characteristicUUID": "00002a04-0000-1000-8000-00805f9b34fb",
                    "byteFrom": 0,
                    "byteTo": -1
                },
                {
                    "key": "Central Address Resolution",
                    "method": "read",
                    "characteristicUUID": "00002aa6-0000-1000-8000-00805f9b34fb",
                    "byteFrom": 0,
                    "byteTo": -1
                }
            ],
            "attributeUpdates": [
            ],
            "serverSideRpc": [
            ]
        }
    ]
}
```

**Connector name:**
BLE Connector

**Versions:**
 - OS: Raspbian GNU/Linux 10
 - Thingsboard IoT Gateway version 2.4.0
 - Python version 3.7.3

Extulen, 2020-08-18:
Hello, I have yet to solve this issue. Is this BLE connector not functional in thingsboard-gateway or is it a bad configuration file?

Extulen, 2020-08-19:
Same problem in thingsboard-gateway 2.5.0

antaresf, 2021-08-13:
in my case solved it by changing the letter in characteristicUUID to capital

samson0v, 2021-12-09:
Hi @Extulen and @antaresf, we are remaking the BLE connector now. We will let you know when it is done.

Thanks for your interest in ThignsBoard IoT Gateway!

samson0v, 2021-12-17:
Hi @Extulen and @antaresf, we did a remake of the BLE Connector yesterday, so you can install the new version of Gateway from the master branch and test the connector.
Let us know if the new version of BLE Connector fixes your problem.

closed 2022-01-05 by zbeacon