thingsboard/thingsboard-gateway #373: [BUG] Python3-paho-mqtt dependency issue

Lautaro-a, 2020-08-13:
I've recently been trying to install Thingsboard Gateway on a gateway running linux Debian 9 32bits, but the .deb package installer says:

``Couldn't satisfy dependency: python3-paho-mqtt``

so i tried:

```
sudo pip3 install paho-mqtt
```

Then i get the response
``
Requirement already satisfied in /usr/local/lib/python3.5/dist-packages (1.5.0).
``

I've also tried with
```
sudo apt-get install -y python3-paho-mqtt
```
but the response is
``
Couldn't find the package python3-paho-mqtt
``

 - OS: Debian GNU/ Linux 9 32bits
 - Thingsboard IoT Gateway version 2.5.0- 1
 - Python3.5.3 (outcome of `python --version` , i also have 2.7 installed)

zbeacon, 2020-08-18:
Hi @Lautaro-a 

Thank you for your interest in ThingsBoard IoT gateway. 
Your issue looks like a problem with apt repository, apt could not find the package in it. to solve it you can try to install the required package for thingsboard_gateway user, using the following command:
`sudo pip3 install paho-mqtt --user thingsboard_gateway`. If this action doesn't solve the problem please let me know and I will move this dependency to install it later, on gateway start from pip.

Lautaro-a, 2020-08-18:
Hi @zbeacon , thanks for your attention, after using the command ```sudo pip3 install paho-mqtt --user thingsboard_gateway``` , i get the outcome that all requeriments are already satisfied, but the package installer still says:
```Couldn't satisfy dependency: python3-paho-mqtt```

zbeacon, 2020-10-22:
Hi @Lautaro-a ,

Currently this issue sometimes appears on some systems. Unfortunately currently, we don't know why it causes the dependency issue. The workaroud described here #378 . Please use it, we are investigating this issue.

closed 2020-12-02 by zbeacon