thingsboard/thingsboard-gateway #374: MQTT messages are terribly slow

rteijeiro, 2020-08-13:
Every 0.5 seconds we are sending 13 messages over MQTT to the Thingsboard Gateway, containing roughly 70 data points. 

These are then expected to the propagated to the Thingsboard application where it can be displayed, however, the time to process one message is extremely long.

I have verified that all messages are sent from the local MQTT client, and the broker handles that update frequency just fine, but it seems like the MQTT Connector cannot handle it.

I have tracked the longest delays to the convert method in the json_mqtt_uplink_converter.py

Any idea what could possibly be wrong?

Version: Thingsboard v.3.0.1PE (Let me know if you need me to provide more information about the configuration or server specs)

XShoreSoftware, 2020-08-28:
We are also experiencing this problem.

zbeacon, 2020-09-01:
Hi @rteijeiro 

Thank you for your interest in ThingsBoard IoT gateway. The gateway was developed to duplicate integration functionality. But it has lower speed of data processing. If you use the PE version of the ThingsBoard we recommend to use Integration instead of the gateway using.

closed 2020-09-01 by zbeacon

XShoreSoftware, 2020-09-03:
This is really not great news to hear. One of the reasons for choosing Thingsboard is the gateway connectors and quite prominent on your guides, now to hear that it's not really to be used for production adds a lot of complexity to the Thingsboard 'offering' and will make us think twice about using this platform in the future.

tolap22, 2025-03-10:
I came across this post because we're experiencing similar performance issues with the gateway. I recently fixed a related bug in the gateway connector, which has been submitted to your codebase. While it now functions correctly, it still suffers from performance problems.
Hearing (5 years later) that the gateway connectors may still not be suitable for production, despite being a key selling point in your guides, only adds to our concerns. It significantly increases the complexity of adopting ThingsBoard.