thingsboard/thingsboard-gateway #375: [BUG]

ttsmousavi, 2020-08-14:
**Describe the bug**
From dashBoard, the user cannot update (set value) attributes that is not a string.
DASHBOARD -> Create "Input widgets" -> "Update shared Integer attribute"

GW throw an exception 

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
```
replace() argument 2 must be str, not bool
/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py Line:539 and 532

Solution: replacing  <content["data"][attribute_key]> to <str(content["data"][attribute_key])>
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - Thingsboard IoT Gateway version [2.5.0]
 - Python version[2.7.17]

ExPS-Pisko, 2020-08-20:
Hi ttsmousavi, I have experienced the same problem. The thing is that the INT type in Thingsboard is a LongINT (64-bit). Check in your OPC server (for example using the UA Expert or some tool from Matrikon) the datatype of the variable you want to write into. Once I changed from INT to this LongINT in my PLC, the write is OK without failure.

The same with other data-types (Real, String...)

Best regards Michal

zbeacon, 2020-09-01:
Hi @ttsmousavi ,

The issue was with message to the external broker formatting. Please try to use the version from the master branch.

closed 2020-09-11 by zbeacon