thingsboard/thingsboard-gateway #378: [BUG] Gateway installation failed - Depends: python3-paho-mqtt but it is not installable

DanielRichart, 2020-08-17:
**Describe the bug**
I'm trying to install the python thingsboard gateway on my ubuntu machine.

When i run `sudo apt install ./python3-thingsboard-gateway.deb -y`, the installation fails and print:

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  python3-thingsboard-gateway : Depends: python3-paho-mqtt but it is not installable
> E: Unable to correct problems, you have held broken packages.

It says that the installation depends on paho-mqtt, but I have the package installed. 

Please, how can I complete the installation?

Thank you,
Daniel Richart

zbeacon, 2020-08-18:
Hi @DanielRichart ,

Thank you for your interest in ThingsBoard IoT gateway.

Please try to install the required package for the thingsboard_gateway user using the following command:
`sudo pip3 install paho-mqtt --user thingsboard_gateway`

And run the installation command again.

DanielRichart, 2020-08-20:
Hi,

Thanks for the answer!

I have tried it, but the error still occurs. There are other things I can try?

Thank you,
Daniel

thucar, 2020-09-08:
Facing the same issue. I have paho-mqtt installed using pip3 but since dpkg seems to be looking for the actual .deb, it still complains about the missing dependency

thucar, 2020-09-08:
A workaround:

manually install all packages and python modules mentioned in the tb-gateway installation guide. Then install the downloaded .deb file using:

`sudo dpkg -i --ignore-depends=python3-paho-mqtt python3-thingsboard-gateway.deb`

closed 2020-10-22 by zbeacon