thingsboard/thingsboard-gateway #381: OPC UA - Update Shared Attributes in TB once changed in PLC

ExPS-Pisko, 2020-08-20:
Hi folks, I am using PLC Siemens - OPC UA Server from Siemens - TB Gateway (Python) - TB 3.0

I need to write some parameter down to PLC. I use the Shared attributes in TB and through the Gateway, the parameter write functions perfectly. However, when this parameter changes in PLC (NOT from TB!!!), the shared attribute does not update in TB...

Does anybody know how to do that? Am I missing something in the New TB Gateway philosophy?

OK, I can read the value from PLC as Client Side attribue and have a rule to change also the Shared attribute (is it not endless loop???) but this seems to me very cumbersome...

Any suggestions are welcome! Best regards Michal

zbeacon, 2020-08-20:
Hi @ExPS-Pisko 

Thank you for your interest in ThingsBoard IoT gateway.

The gateway, like any device in ThingsBoard cannot change shared attributes, it can only subscribe and get their values.
More information [here](https://thingsboard.io/docs/user-guide/attributes/#attribute-types).

ExPS-Pisko, 2020-08-20:
Dear Mr. Barkov, and is it possible to write from TB (through TB Gateway) to the Client Attributes (located in OPC Server)?

 

With the old Java gateway, I was able to write using the RPC command „Write“ + params. In the new GW, I was not able to set this to work…

 

Thanks and best regards

Ing. Michal Pisko, Ph.D. 
  
Managing director 
ExPS s.r.o. - Office Slany 
Ouvalova 390, 274 01 Slany

Czech Republic 
  
Tel. +420 312 520 507 
Mob. +420 732 608 300 
E-mail:  <mailto:michal.pisko@exps.cz> michal.pisko@exps.cz 
 <http://www.exps.cz/> http://www.exps.cz

closed 2020-09-01 by zbeacon