thingsboard/thingsboard-gateway #382: [HELP] Tb updater error

sobehard, 2020-08-21:
**Describe the issue**
There is an issue from the tb update progress. The issue did not occur immediately when the thingsboard gateway start or restart, it only happen when the thingsboard gateway already run for some time.

**Error traceback (If it was raised):**
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 601, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 346, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 852, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 284, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x7f10ec019978>: Failed to establish a new connection: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 440, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 639, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='updates.thingsboard.io', port=443): Max retries exceeded with url: /api/tb-gateway/updates (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f10ec019978>: Failed to establish a new connection: [Errno -2] Name or service not known',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_updater.py", line 72, in check_for_new_version
    response = post(**request_args)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 112, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 58, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 520, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 630, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 508, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='updates.thingsboard.io', port=443): Max retries exceeded with url: /api/tb-gateway/updates (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f10ec019978>: Failed to establish a new connection: [Errno -2] Name or service not known',))

```

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.5.0
 - Python version 3.7

zbeacon, 2020-08-21:
Hi @sobehard,

It is just an error with connection to the update server, there is nothing critical, but I've added handler for this exception to the master branch, you can use it to avoid error trace printing.

sobehard, 2020-08-21:
Alright, thanks for fast response. I will closed this issue.

closed 2020-08-21 by sobehard