thingsboard/thingsboard-gateway #383: [HELP] MQTT over SSL/TLS issue

sobehard, 2020-08-21:
**Describe the issue**
There is an issue related to the mqtt over ssl/tls which make the thingsboard gateway become inactive and stop converting mqtt message from local sensor to the cloud (thingsboard). At first, everything is working fine without any issue. After some time, the issue will come out and I need to force restart the thingsboard gateway in order to receive data from local sensor. I'm not sure the exact timing and the thing that caused this issue. 

**tb_gateway.yml file setting** 
[image]

**Error traceback (If it was raised):**
```
""2020-08-21 11:51:57" - ERROR - [tb_client.py] - tb_client - 129 - SSL/TLS has already been configured."
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 125, in run
    min_reconnect_delay=self.__min_reconnect_delay)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 162, in connect
    ciphers=None)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 709, in tls_set
    self.tls_set_context(context)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 618, in tls_set_context
    raise ValueError('SSL/TLS has already been configured.')
ValueError: SSL/TLS has already been configured.
""2020-08-21 11:51:58" - ERROR - [tb_client.py] - tb_client - 129 - SSL/TLS has already been configured."
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 125, in run
    min_reconnect_delay=self.__min_reconnect_delay)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 162, in connect
    ciphers=None)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 709, in tls_set
    self.tls_set_context(context)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 618, in tls_set_context
    raise ValueError('SSL/TLS has already been configured.')
ValueError: SSL/TLS has already been configured.
""2020-08-21 11:51:59" - ERROR - [tb_client.py] - tb_client - 129 - SSL/TLS has already been configured."
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 125, in run
    min_reconnect_delay=self.__min_reconnect_delay)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 162, in connect
    ciphers=None)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 709, in tls_set
    self.tls_set_context(context)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 618, in tls_set_context
    raise ValueError('SSL/TLS has already been configured.')
ValueError: SSL/TLS has already been configured.
""2020-08-21 11:52:00" - ERROR - [tb_client.py] - tb_client - 129 - SSL/TLS has already been configured."
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 125, in run
    min_reconnect_delay=self.__min_reconnect_delay)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 162, in connect
    ciphers=None)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 709, in tls_set
    self.tls_set_context(context)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 618, in tls_set_context
    raise ValueError('SSL/TLS has already been configured.')
ValueError: SSL/TLS has already been configured.

```

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.3.0 and 2.5.0
 - Python version 3.7

zbeacon, 2020-08-21:
Hi @sobehard ,

Thank you for your feedback, please try to use the version from the master branch of the thingsboard-gateway repository.

sobehard, 2020-08-21:
Hi @zbeacon ,

I just downloaded the last code from master branch and test the ssl/tls connection. There is another issue occur when I try to run the thingsboard-gateway. Below is the error traceback from the log.

```
Traceback (most recent call last):
  File "tb_gateway.py", line 31, in <module>
    main()
  File "tb_gateway.py", line 22, in main
    TBGatewayService(path.dirname(path.abspath(__file__)) + '/config/tb_gateway.yaml'.replace('/', path.sep))
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.5.0-py3.6.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 87, in __init__
    self.tb_client = TBClient(config["thingsboard"])
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.5.0-py3.6.egg/thingsboard_gateway/gateway/tb_client.py", line 53, in __init__
    self.client._client.tls_insecure_set(False)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 743, in tls_insecure_set
    self._ssl_context.check_hostname = not value
ValueError: check_hostname needs a SSL context with either CERT_OPTIONAL or CERT_REQUIRED
^CException ignored in: <module 'threading' from '/usr/lib/python3.6/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
```

zbeacon, 2020-08-21:
Ok, please try again the version from master branch.

sobehard, 2020-08-21:
[image]
I updated the thingsboard-gateway and try the latest gatewa code, but he same issue still appear. Below is the screenshot of the log.
[image]

zbeacon, 2020-08-21:
It looks like it wasn't updated, please try to pull again.

sobehard, 2020-08-21:
[image]
Above screenshot show the git pull log.

The issue still appear after second update. 
[image]

zbeacon, 2020-08-21:
Have you removed the old version before installation?

Commands for removing and installation are:

sudo pip3 uninstall thingsboard-gateway
pip3 uninstall thingsboard-gateway

python3 setup.py install

sobehard, 2020-08-21:
Oh sorry, you're right. I'm using wrong command (sudo apt remove python3-thingsboard-gateway) to uninstall the thingsboard-gateway just now. The thingsboard gateway is able to run now. 
[image]

closed 2020-09-01 by zbeacon