thingsboard/thingsboard-gateway #385: [BUG]

seaowl, 2020-08-27:
Using all-digits characters in request.json URL, rises an error.

Steps to reproduce: 

1. Enable "request connector" in thingsboard_gateway/config/tb_gateway.yaml
2. Edit thingsboard_gateway/config/request.json, change `"url": "getdata"` to `"url": "3100"`
3. Run tb_gateway.py

**Connector name:** request connector

**Error traceback:**
```
""2020-08-27 17:23:11" - ERROR - [request_connector.py] - request_connector - 211 - Extra data: line 1 column 2 - line 1 column 22 (char 1 - 21)"
Traceback (most recent call last):
  File "C:\dev\pycharm\projects\thingsboard-gateway\thingsboard_gateway\connectors\request\request_connector.py", line 195, in __send_request
    data_to_storage.append(response.json())
  File "C:\dev\python\python385\lib\site-packages\requests\models.py", line 898, in json
    return complexjson.loads(self.text, **kwargs)
  File "C:\dev\python\python385\lib\site-packages\simplejson\__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "C:\dev\python\python385\lib\site-packages\simplejson\decoder.py", line 373, in decode
    raise JSONDecodeError("Extra data", s, end, len(s))
simplejson.errors.JSONDecodeError: Extra data: line 1 column 2 - line 1 column 22 (char 1 - 21)

Process finished with exit code -1
```

**Versions:**
 - OS: Windows 10 x64
 - Thingsboard IoT Gateway version 2.5.0
 - Python 3.8.5

seaowl, 2020-08-27:
upd: this is not related to "all-digits in URL", but rather specific reply from device rises the error.

zbeacon, 2020-09-01:
Hi @seaowl 

Thank you for your interest in ThingsBoard IoT gateway and feedback. Please try to use the version from the master branch.

closed 2020-09-25 by zbeacon