thingsboard/thingsboard-gateway #386: [HELP] Write to closed file

Tikarei, 2020-09-02:
**Describe the issue**
Getting I/O error when using file storage. Do I need to increase the file count or something else?

**Configuration** (Attach your configuration file)
```
thingsboard:
  host: tbce_1
  port: 1883
  remoteConfiguration: true
  security:
    accessToken: token
storage:
  type: file
  data_folder_path: ./data/
  max_file_count: 10
  max_read_records_count: 1000
  max_records_per_file: 100000
connectors:
  -
    name: OPC-UA Connector AkerBP
    type: opcua
    configuration: opcua.json
```

**Connector name:**
OPCUA

**Error traceback (If it was raised):**
```
""2020-09-02 08:47:41" - ERROR - [file_event_storage.py] - file_event_storage - 41 - write to closed file"
Traceback (most recent call last: File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.4.1-py3.7.egg/thingsboard_gateway/storage/file_event_storage.py", line 37, in put self.__writer.write(event) 
File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.4.1-py3.7.egg/thingsboard_gateway/storage/event_storage_writer.py", line 58, in write self.buffered_writer.write(encoded + linesep.encode('utf-8')) 
ValueError: write to closed file
""2020-09-02 08:47:41" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 341 - Data from the device "23-FT-92512" cannot be saved, connector name is OPC-UA Connector."
```

**Versions (please complete the following information):**
 - OS: docker host is Ubuntu 18.04
 - Thingsboard IoT Gateway version: latest docker image 2.4.1
 - Python version: 3.7, whichever comes with the docker image

zbeacon, 2020-09-09:
Hi @Tikarei 

Thank you for your feedback, I made some changes in code, could you try to run the version from develop/2.4-python branch?

Tikarei, 2020-09-09:
Yes, I will do this as soon as I have some time and let you know the results.

Tikarei, 2020-09-11:
@zbeacon I have not received the error after using the new version as of yesterday morning.

zbeacon, 2020-09-11:
@Tikarei , it is nice, I close this issue, but if this situation repeats write the new one please.

commit 20fc093f (2020-09-09): Fix #386
closed 2020-09-11 by zbeacon