thingsboard/thingsboard-gateway #387: [HELP] IoT Gateway MQTT Connector [Errno 99] Cannot assign requested address

felixmondelo, 2020-09-07:
**Describe the issue**
I have configured an IoT Gateway on the Thingsboard portal, also I have installed a Gateway on a remote server. 

On Gateway server, I have installed Mosquitto that is working ok. I configured tb_gateway.yaml with remote Thingsboard server and if I don't configure any connector, it connects perfect to the remote Thingsboard server.

Now I want to configure MQTT connector with my Mosquitto Broker, but when I start the gateway I get this error:

ERROR - [mqtt_connector.py] - mqtt_connector - 172 - [Errno 99] Cannot assign requested address"

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/5183750/tb_gateway.yaml.txt)
[mqtt.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/5183751/mqtt.json.txt)

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Error traceback (If it was raised):**
```
""2020-09-07 15:03:26" - INFO - [mqtt_connector.py] - mqtt_connector - 203 - MQTT Broker Connector has been stopped."
""2020-09-07 15:03:26" - INFO - [tb_gateway_service.py] - tb_gateway_service - 139 - Gateway started."
""2020-09-07 15:03:28" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 195 - connection SUCCESS"
""2020-09-07 15:03:28" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 2"
""2020-09-07 15:08:54" - INFO - [tb_gateway_service.py] - tb_gateway_service - 76 - Gateway starting..."
""2020-09-07 15:08:54" - INFO - [tb_gateway_service.py] - tb_gateway_service - 81 - ThingsBoard IoT gateway version: 2.4.1"
""2020-09-07 15:08:54" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 1"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted mapping handlers: 3"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected mapping handlers: 0"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted serverSideRpc handlers: 2"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected serverSideRpc handlers: 0"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted connectRequests handlers: 2"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected connectRequests handlers: 0"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted disconnectRequests handlers: 2"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected disconnectRequests handlers: 0"
""2020-09-07 15:08:54" - ERROR - [mqtt_connector.py] - mqtt_connector - 130 - 'attributeRequests' section missing from configuration"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted attributeUpdates handlers: 1"
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected attributeUpdates handlers: 0"
""2020-09-07 15:08:54" - ERROR - [mqtt_connector.py] - mqtt_connector - 172 - [Errno 99] Cannot assign requested address"
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.4.1-py3.7.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 170, in run
    self.__connect()
  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.4.1-py3.7.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 188, in __connect
    self.__broker.get('port', 1883))
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.0-py3.7.egg/paho/mqtt/client.py", line 937, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.0-py3.7.egg/paho/mqtt/client.py", line 1071, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.0-py3.7.egg/paho/mqtt/client.py", line 3522, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/local/lib/python3.7/socket.py", line 728, in create_connection
    raise err
  File "/usr/local/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
OSError: [Errno 99] Cannot assign requested address
""2020-09-07 15:08:54" - INFO - [mqtt_connector.py] - mqtt_connector - 203 - MQTT Broker Connector has been stopped."
""2020-09-07 15:08:54" - INFO - [tb_gateway_service.py] - tb_gateway_service - 139 - Gateway started."
""2020-09-07 15:08:54" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 195 - connection SUCCESS"
""2020-09-07 15:08:54" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 2"
```

**Versions (please complete the following information):**
 - OS: Windows 10
 - Thingsboard IoT Gateway version 2.4.1

zbeacon, 2020-09-08:
Hi @felixmondelo,

It looks like the issue with the addressing (MQTT client in the gateway works with IPv4, but by default the Windows 10 uses IPv6). To check the issue, you can run the following command:
ping localhost in the command line. Probably you will see something like:
`Reply from ::1: time<1ms` - it means that you have IPv6 higher priority than IPv4.

To sole this issue - you should set the value of the host (in MQTT configuration json file) to 127.0.0.1 

Please try and let me know about the results.

felixmondelo, 2020-09-08:
Hi @zbeacon ,

I have changed in mqtt.json the host value to 127.0.0.1, but the problem is the same:  [Errno 99] Cannot assign requested address

Thanks and regards.

felixmondelo, 2020-09-08:
Hi @zbeacon, 

Problem resolved, I assign IPv4 address. In command line, with ipconfig command, I get the IPv4 address:

   IPv4 Address. . . . . . . . . . . : 10.1.6.5

And now it works!!

Thanks and regards.

zbeacon, 2020-09-08:
Thanks for the information, we will consider it to solve issues like this.

closed 2020-09-08 by zbeacon