thingsboard/thingsboard-gateway #388: [HELP] Reading Input Register in Modbus TCP to Thingsboard Gateway

adriortega, 2020-09-10:
Hellow, I am trying to read Input Registers from a Modbus TCP map, using the modbus_TCP.json file.  In order to know the state and the attributes of some of my devices, I want to read single (boolean) or multiple bits variables, like in the following examples.

**Reading a boolean variable: Thats my modbus.json code inside "attributes" of the Device.**

{
            "tag": "FlagSystemOk",
            "functionCode": 4,
            "type": "bits",
            "objectsCount": 1,
            "address": 6,
            "bit":15            
          },

**Result in Thingsboard:** 

2020-09-10 16:18:54	FlagSystemOk	      [false,false,false,false,false,false,false,false]

**Question**: Why does the telemetry shows this 8 bit register, instead of only one or the value of the bit?

Regards and thanks for your answers.

Adrian.

*Versions (please complete the following information):**
 - OS: Windows 7
 - Thingsboard IoT Gateway version [2.5.0]
 - Python version [3.7.8]

zbeacon, 2020-09-11:
Hi @adriortega ,

Thank you for your feedback, I have fixed it in the develop/2.4-python branch please try to use it and let me know about the results.

adriortega, 2020-09-11:
> Hi @adriortega ,
> 
> Thank you for your feedback, I have fixed it in the develop/2.4-python branch please try to use it and let me know about the results.

adriortega, 2020-09-11:
Hi again @zbeacon . I am very pleased for your fast response and help. I have just used your recently fixed version and I still have some problems in this issue.  In the next image you can see the error messages in my command window.

[image]

It seems as the range of the register is overpassed when I only tried to read the 15th bit of a 16 bits register.

Thanks for your previous help, and I hope that you know how can I solve this new problem.
best whises.

Adrian.

zbeacon, 2020-09-11:
Please try to set up the value objectsCount to 2 instead 1 in config.

adriortega, 2020-09-11:
Hi! Lamentably the error continues. I attach here the command prompt with the messages.

[image]

Thanks for your help, and I have also another question, can I read 2 o 3 or less than 8 bits of an input register instead of reading the entire register and then decoding it? If so, can you tell me how??

I have tried to do it in the following way:

{
            "tag": "MainState",
            "type": "bits",
            "functionCode": 4,
            "objectsCount": 2,
            "address": 1 
          },

Thank you very much.

Regards.

Adrian.

closed 2021-11-05 by zbeacon