thingsboard/thingsboard-gateway #390: [BUG] file storage reader reads partially line (concurrency problem?)

Chabrol, 2020-09-11:
**Describe the bug**
Sometimes I see log entries like:
```
ERROR - event_storage_reader - 56 - Incorrect padding'
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/storage/event_storage_reader.py", line 51, in read
    self.current_batch.append(b64decode(line).decode("utf-8"))
  File "/usr/lib/python3.5/base64.py", line 88, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
```
I suppose the reason is an concurrent file access from file reader and writer (storage). The system is fairly loaded with about 50 messages per second. Additionally it looks like the record is lost.

**Versions (please complete the following information):**
 - OS: Debian 9.13
 - ThingsBoard IoT gateway version: 2.5.0
 - Python 3.5.3

zbeacon, 2020-09-14:
Hi @Chabrol,

Thank you for your feedback. It looks like you try to save some data in unicode or some else coding, the gateway can save data  in UTF-8. Please check what do you receives after converter. It looks like some data cannot be converted to UTF-8. To investigate it deeply please also specify which connector you used and your configuration for it.

Chabrol, 2020-09-14:
I put some debugging output to the storage reader. It sometimes gets incomplete base64 encoded lines, which of course can't be decoded. If you check the data file in such a case, the line is intact and I can decode it with the linux command line tool base64. Therefore the problem is at the lower file layer and has nothing to do with the interpretation of the decoded base64 payload.

samson0v, 2021-08-02:
Hi Chabrol, please, update your Thingsboard Gateway to the newest version via master branch.

closed 2021-09-06 by zbeacon