thingsboard/thingsboard-gateway #392: [HELP]

ahweeuhl, 2020-09-14:
**Describe the issue**
Modbus TCP/IP connector is being used. It shows that modbus connector is connected and Thingsboard Gateway is also connected. My device sees the correct function code and register request and it does not give any errors. But I don't see any Telemetry data going to Thingsboard.

I only see events produced but that's it in the telemetry data. 

I also used Modbus-poll(third party modbus master)l to double check and I get the data on that. 

**Connector name (If you need help with some connector/converter):**
Modbus TCP
the connector log shows that it starts, connects, then stops. 
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/5216291/connector.log)

[tb_connection.log](https://github.com/thingsboard/thingsboard-gateway/files/5216297/tb_connection.log)

● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-09-13 22:07:59 PDT; 10min ago
 Main PID: 3340 (python3)
    Tasks: 7 (limit: 2065)
   CGroup: /system.slice/thingsboard-gateway.service
           └─3340 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Sep 13 22:07:59 raspberrypi systemd[1]: Started ThingsBoard Gateway.
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [tb_gateway_service.py] - tb_gateway_service - 72 - Gateway starting..."
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [tb_gateway_service.py] - tb_gateway_service - 77 - ThingsBoard IoT gateway version: 2.5.0"
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 1"
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [modbus_connector.py] - modbus_connector - 68 - Starting Modbus connector"
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [modbus_connector.py] - modbus_connector - 74 - Modbus connected."
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [tb_gateway_service.py] - tb_gateway_service - 141 - Gateway started."
Sep 13 22:08:01 raspberrypi python3[3340]: ""2020-09-13 22:08:01" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
Sep 13 22:13:51 raspberrypi python3[3340]: ""2020-09-13 22:13:51" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"

**Versions (please complete the following information):**
 - OS: [RASPBIAN BUSTER]
 - Thingsboard IoT Gateway version [2.5]
 - Python version[e.g. 3.7]
-Thingsboard Professional Edition
[tb_gateway_yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/5216310/tb_gateway_yaml.txt)
[modbusconnector.txt](https://github.com/thingsboard/thingsboard-gateway/files/5216314/modbusconnector.txt)

zbeacon, 2020-09-14:
Hi @ahweeuhl 

Please set up the logging level from INFO to DEBUG in logs.conf, collect logs and send them.

ahweeuhl, 2020-09-15:
Here is the logs with DEBUG enabled. 
i had to condense it as I can't upload over 10mb

[condensedConnector.txt](https://github.com/thingsboard/thingsboard-gateway/files/5222529/condensedConnector.txt)
[condensedConverter.txt](https://github.com/thingsboard/thingsboard-gateway/files/5222530/condensedConverter.txt)

[storage.log](https://github.com/thingsboard/thingsboard-gateway/files/5222533/storage.log)
[tb_connection.log](https://github.com/thingsboard/thingsboard-gateway/files/5222534/tb_connection.log)
[service.log](https://github.com/thingsboard/thingsboard-gateway/files/5222535/service.log)

zbeacon, 2020-09-15:
Hi @ahweeuhl ,

Where are you looking for the telemetry, it looks like the gateway save the telemetry to itself, of course if your gateway is named piGateway on ThingsBoard. Please try to change deviceName to another in the modbus configuration file and restart the gateway.

ahweeuhl, 2020-09-15:
I'm trying to look at the telemetry on Thingsboard. I renamed the device to "SERVODRIVE" instead of piGateway and it doesn't seem to get on Thingsboard. just also an fyi, using the raspberry pi which thingsboard gateway is installed on, and to poll the modbus device. 

And to make sure I understand, the deviceName is what is on Thingsboard to be also named the same in the modbus config file. What about the "name" of the server that is in the modbus config file? 

i doubled checked the port is open, and used mqtttest to make sure everything is good and i'm getting telemetry from that test. I'm puzzled where it's going.

ahweeuhl, 2020-09-16:
I didn't check the "is gateway" on Thingsboard I can't believe it (face palm). Maybe some edits on the docs to make sure "is Gateway" checked off.

ahweeuhl, 2020-09-16:
it's all working now after checking the box. Thank you. i'll close it.

closed 2020-09-16 by ahweeuhl