thingsboard/thingsboard-gateway #393: [HELP] Method to modify the source code to disable gateway telemetry (or publish in longer period)

sobehard, 2020-09-14:
Hi,

May I know how to modify the source to so that I can remove/put longer delay for the gateway telemetry part in the tb_gateway_service.py file? I had tried to modify some line in the python file, but it doesn't change the output. 

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04]
 - Thingsboard IoT Gateway version 2.5.0
 - Python version 3.7

zbeacon, 2020-09-14:
Hi @sobehard ,

To run you gateway with changes, you should follow next steps:
1. Make changes in the source code.
2. Remove previous version of the gateway from your system using command `pip3 uninstall thingsboard-gateway`
3. Install the new version with your changes using command from root folder of repository `python3 setup.py install`
4. Run the gateway using command `python3 thingsboard_gateway/tb_gateway.py` from the root folder of the repository.

If you want to change the daemon - remove and install new library for user `thingsboard_gateway`, you can do this just add `--user thingsboard_gateway` to remove or install command.

sobehard, 2020-09-14:
Hi @zbeacon ,

Alright, thanks for the assist and fast response. I will closed the topic right now.

closed 2020-09-14 by sobehard