thingsboard/thingsboard-gateway #396: Not able to update gateway programatically

shirishareddy, 2020-09-21:
I am using TB gateway software in one of my projects.

I am now implementing OTA update of Thingsboard gateway.

Here are the steps I follow.
1) Build 2.5.0 as debain package and install it using **sudo apt install 'debiab package' -y**
2) 2.5.0 contains the functionality to update the gateway software using tar.gz file from specific location in my device. Below is command i am using in program.
**check_call([executable, "-m",  "pip", "install", archivePath, "--upgrade"])**
3) I built tar.gz file by setting version as 2.6.0 in setup.py file.

**Issue**:  the update code in 2.5.0 is getting executed and installing the new software, but it is getting installed in /var/lib/thingsboard-gateway directory instead of /usr/local/lib. Hence get_distribution is returning version as 2.5.0. Any ideas on how can we avoid the installtion in /var/lib/thingsboard-gateway folder?

**Note:** when I install new tar.gz file from command line, the installation works fine and get_distribution is returning 2.6.0. I am not able to figure out the issue with programmatic installation.

**Versions:**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.5.0.
 - Python 3.6

shirishareddy, 2020-09-22:
The same issue is observed when upgrading TB gateway software from 2.3.0 to 2.3.1.
Did anyone from your team verified TBUpdater functionality?

zbeacon, 2020-09-22:
Hi @shirishareddy ,

>> 2.5.0 contains the functionality to update the gateway software using tar.gz file from specific location in my device. Below is command i am using in program.
>> check_call([executable, "-m", "pip", "install", archivePath, "--upgrade"])

The gateway doesn't contain this code inside. And the gateway doesn't support the update from tar.gz file. It is able to install updates directly from pip. This functionality works. Please share the link to official site where you have read about update from tar.gz file?

Anyway, even if you have added this functionality. The issue can be caused by user from daemon file. because the gateway runs as thingsboard_gateway user. Probably you try to install package as root user or using sudo. To install it in thingsboard_gateway environment you can add `--user thingsboard_gateway` in your installation command, something like `check_call([executable, "-m", "pip", "install", archivePath, "--upgrade","--user","thingsboard_gateway"])`

shirishareddy, 2020-09-22:
Hi Illia Barkov,

 Thanks for the reply.

zbeacon, 2020-09-22:
Ok, I understood, what you mean. 

Could you try to check the gateway version using RPC request to gateway command `gateway_version` will it show the correct version? If no please try to re-run gateway service using systemctl or RPC request `gateway_restart 0` ands check again the version.

shirishareddy, 2020-09-22:
> Ok, I understood, what you mean.
> 
> Could you try to check the gateway version using RPC request to gateway command `gateway_version` will it show the correct version? If no please try to re-run gateway service using systemctl or RPC request `gateway_restart 0` ands check again the version.

I restarted daemon using systemctl but still I get the old version

zbeacon, 2020-09-22:
Did you check using gateway_version rpc command? update command uses pip to update the version of the gateway. It doesn't override the version in the debian package. Anyway I will try to reproduce it and write here.

shirishareddy, 2020-09-22:
I didn't check using gateway_version rpc command. But from the command line I checked using this function.
get_distribution("thingsboard_gateway").version

Though 2.3.1 is installed the above function gives the version as 2.3.0
Note: 2.3.1 is installed in /var/lib/thingsboard_gateway location.

closed 2021-01-18 by zbeacon