thingsboard/thingsboard-gateway #397: [HELP] OPC UA deviceNodePattern

yeduy, 2020-09-24:
Hello,

I am having trouble typing the correct node path. Spaces and dots are used when creating the names of the folders. How can write this node path?

-Objects
   -Matrikon.OPC.Simulation(DA)
      -Simulation Items
         -Random

`{
  "server": {
    "name": "OPC-UA Server",
    "url": "192.168.70.43:48403",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Matrikon.OPC.Simulation(DA)\\.Simulation Items\\.Random",
        "deviceNamePattern": "Root\\.Objects\\.Matrikon.OPC.Simulation(DA)\\.Simulation Items\\.Random",
        "attributes": [
          {
            "key": "boolean",
            "path": "Boolean"
          }
        ],
        "timeseries": [
          {
            "key": "money",
            "path": "Money"
          },
          {
            "key": "real4",
            "path": "Real4"
          }
        ]
      }
    ]
  }
}
`

` thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-09-24 11:46:32 MSK; 6s ago
  Process: 4835 ExecStop=/bin/kill -INT $MAINPID (code=exited, status=0/SUCCESS)
 Main PID: 4836 (python3)
    Tasks: 9 (limit: 4915)
   CGroup: /system.slice/thingsboard-gateway.service
           └─4836 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Sep 24 11:46:34 basari python3[4836]:     val = from_binary(uatype, data)
Sep 24 11:46:34 basari python3[4836]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 482, in from_binary
Sep 24 11:46:34 basari python3[4836]:     return unpack_uatype(vtype, data)
Sep 24 11:46:34 basari python3[4836]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 200, in unpack_uatype
Sep 24 11:46:34 basari python3[4836]:     return st.unpack(data)
Sep 24 11:46:34 basari python3[4836]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 138, in unpack
Sep 24 11:46:34 basari python3[4836]:     return struct.unpack(self.format, data.read(self.size))[0]
Sep 24 11:46:34 basari python3[4836]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/opcua/common/utils.py", line 65, in read
Sep 24 11:46:34 basari python3[4836]:     raise NotEnoughData("Not enough data left in buffer, request for {0}, we have {1}".format(size, self))
Sep 24 11:46:34 basari python3[4836]: opcua.common.utils.NotEnoughData: Not enough data left in buffer, request for 4, we have Buffer(size:0, data:b'')`

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

ExPS-Pisko, 2020-10-07:
What is the exact configuration of Matrikon you use? Where to download this version? The thing is that I am not aware of matrikon simulation server having the OPC UA. I think it has only the OPC classic... Do you use the UA Wrapper on top of the Simulation server?
Once this is clarified, we can move on to the configuration of python. For example, this is my connection string to data in siemens PLC: 
"mapping": [
       {
        "deviceNodePattern": "ns=3;s=\"Flow_Controller_DB\"",
        "deviceNamePattern": "FC1581",
        "attributes": [
          {
            "key": "MV_HMI",
            "path": "ns=3;s=\"Flow_Controller_DB\".\"MV_INT\""
          },
          {
            "key": "SP_HMI",
            "path": "ns=3;s=\"Flow_Controller_DB\".\"SP_INT\""
          }
        ],
        "timeseries": [],
        "rpc_methods": [],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "Download_MV",
            "attributeOnDevice": "ns=3;s=\"Flow_Controller_DB\".\"MV_INT\""
          }
        ]
      }
    ]

I am running on Windows system

Best regards

ExPS-Pisko, 2020-10-08:
I see that the the "escaping" disappeared. I use the "\" to escape only the inner quotes of the Path string, not the dots.

Will try to get it work (appear in this comment):  "path": "ns=3;s=\\"Flow_Controller_DB\\".\\"MV_INT\\""

zbeacon, 2020-10-08:
Hi @ExPS-Pisko ,

Thank you for your feedback, there was an error in OPC-UA connector, please try the gateway version from master branch and use configuration from your first message here.

yeduy, 2020-10-08:
Hi @ExPS-Pisko ,

Thank you for your answer. Yes, we used OPC DA-UA Wrapper top of the Matrikon Server.

We solve the problem with the help of @zbeacon  , using the gateway version from master branch.

closed 2020-10-08 by yeduy