thingsboard/thingsboard-gateway #401: [HELP] Does the new tb-gateway still support multiple tenants?

thucar, 2020-09-30:
**Describe the issue**
I have thingsboard-gateway running perfectly for single tenant. However I have not found any reference for how to configure gateways for multiple tenants. Old .yaml configuration file had a section called "gateways" for that. Now we only have section "thingsboard" where the security and access token is defined.

agtokty, 2020-09-30:
Hi @thucar , as far as i know, the gateway service is can be simulate only one gateway device at the same time, and a gateway device belongs to only one tenant. you can't configure this scenario currently

thucar, 2020-09-30:
That is unfortunate. @zbeacon Do you know if there are any plans for multi-tenant setup support getting included into this new version of tb-gateway again at some point?

zbeacon, 2020-09-30:
Hi @thucar ,

Thank you for your question. @agtokty is completely right, there is no ability to use multi tenant feature in python gateway. And by now we don't have plans to add it into the project, because this feature needs a lot of changes in architecture of the project. 

But at the same time you can try to run several gateway in the docker containers and connect to different tenants with the same configuration. It can be useful in some cases.

closed 2020-10-22 by zbeacon

sferguson-ISI, 2023-05-12:
@imbeacon can you elaborate on the workaround for multiple tenants?  Can you have a native thingsboard-gateway service running as well as a containerized thingsboard-gateway running on the same machine and connected to the same thingsboard instance?  I just stumbled upon the fact that you thingsboard-gateway doesn't support Thingsboard multi-tenancy.