thingsboard/thingsboard-gateway #404: Twelve position longitude OID problems

anibal8300, 2020-10-05:
**Describe the bug**
When the OID of the equipment has more than 11 positions, it does not send the information to Thingsboard.
For example 1.3.6.1.2.1.31.1.1.1.7.1

[SNMP Connector]

{
  "devices": [
    {
      "deviceName": "SNMP router",
      "deviceType": "snmp",
      "ip": "192.168.88.1",
      "port": 161,
      "pollPeriod": 5000,
      "community": "public",
      "attributes": [
        {
          "key": "ReceivedFromGet",
          "method": "get",
          "oid": "1.3.6.1.2.1.1.2.0",
          "timeout": 6
        }
      ],
      "telemetry": [
        {
          "key": "ReceivedFromGet",
          "method": "get",
          "oid": "1.3.6.1.2.1.31.1.1.1.7.1"
        }
      ]
    }
  ]
}

**Error traceback (If available):**

 - OS: Ubuntu 18.04
 - ThingsBoard IoT gateway version: 2.5.1
 - Python version 3.8

mgrey, 2021-04-15:
Not sure if related, but I can not get the get method to work at all when dealing with integers.
No attribute/telemetry data is received in this case. A tcpdump shows request and response ok.
If I request a string, than it works.
My workaround is to use getnext method that returns a json string: {'<oid>':'<value>} and to parse the value out with rule-chain and then store is as an integer.

JoachimVeulemans, 2021-05-20:
@zbeacon I am having the same problem. Strings are working fine, Integers not. Any why this is not working? Thanks!

JoachimVeulemans, 2021-06-01:
@zbeacon When will there be a new release so I can test your fix?

JoachimVeulemans, 2021-06-07:
@zbeacon I compiled the master branch. This is what I am getting:

`""2021-06-07 04:46:53" - ERROR - [snmp_connector.py] - snmp_connector - 146 - invalid literal for int() with base 10: ''"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/snmp/snmp_connector.py", line 140, in __process_data
    response = self.__process_methods(method, common_parameters, datatype_config)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/snmp/snmp_connector.py", line 156, in __process_methods
    response = puresnmp.get(**common_parameters,
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/puresnmp/api/pythonic.py", line 135, in get
    raw_value = raw.get(ip, community, oid, port, timeout=timeout, version=version)
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/puresnmp/api/raw.py", line 70, in get
    result = multiget(
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/puresnmp/api/raw.py", line 89, in multiget
    parsed_oids = [OID(oid) for oid in oids]
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/puresnmp/api/raw.py", line 89, in <listcomp>
    parsed_oids = [OID(oid) for oid in oids]
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/puresnmp/x690/types.py", line 555, in from_string
    identifiers = [int(ident, 10) for ident in value.split('.')]
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/puresnmp/x690/types.py", line 555, in <listcomp>
    identifiers = [int(ident, 10) for ident in value.split('.')]
ValueError: invalid literal for int() with base 10: ''`

Any ideas?

JoachimVeulemans, 2021-06-28:
@zbeacon With the fix on [issue 526](https://github.com/thingsboard/thingsboard-gateway/issues/526#issuecomment-865790118) I can confirm this issue is resolved and can be closed.

closed 2021-09-06 by zbeacon