thingsboard/thingsboard-gateway #405: TB gateway issue after upgrade from 2.3 to 2.5.1.

zdenoL, 2020-10-05:
Hi, I am getting  after upgrade to 2.5.1
for 2.3.1 it was working.  TLS is still supported as you can see at the connector config?
another question is:
How can I achieve deffinition of  different topic with different mappings in the same broker?

""2020-10-05 09:28:27" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised"
""2020-10-05 09:28:29" - ERROR - [mqtt_connector.py] - mqtt_connector - 193 - [Errno 111] Connection refused"

my mqtt conenctor is set like this:
{
  "broker": {
    "name": "iotplanet_MQTT",
    "host": "mqtt.xxxxx.sk",
    "port": 8883,
    "security": {
      "type": "tls",
      "username": "xxxxx",
      "password": "xxxxx"
    }
  }

ashvayka, 2020-10-07:
Are you sure that the credentials work for your broker? What type of broker do you use?

zdenoL, 2020-10-07:
Hi, it is mosquiotto, and it worked perfectly within version 2.3.1

zbeacon, 2020-10-07:
Hi @zdenoL ,

According to your configuration, the gateway has to use username and password and validate public certificate on MQTT broker. There were some changes in logic for work with TLS since 2.3.1. Could you share credentials for mqtt connection to email ibarkov@thingsboard.io for investigation purposes. Also please check which version of TLS used, the gateway uses version TLS 1.2 to check certificate.

You can add devices into mapping section with different topics.

zdenoL, 2020-10-07:
Hi, so it is not gonna work jsut with this setup but certificate is also required?

"broker": {
"name": "iotplanet_MQTT",
"host": "mqtt.xxxxx.sk",
"port": 8883,
"security": {
"type": "tls",
"username": "xxxxx",
"password": "xxxxx"
}
}

zbeacon, 2021-01-05:
CA certificate (mqttserver.pub.pem) is still required for this type of authorization.