thingsboard/thingsboard-gateway #407: [HELP] Configuration of mqtt.json IoT Gateway Thingsboard

oscardavidtorres1994, 2020-10-06:
I have data comming from TTN, the data seems like this:
```
**{
  "end_device_ids" : {
    "device_id" : "dev4",
    "application_ids" : {
      "application_id" : "machineman5"
    },
    "dev_eui" : "70B3D549990527FA",
    "join_eui" : "70B3D57ED00293DE",
    "dev_addr" : "01C7EB01"
  },
  "correlation_ids" : [ "as:up:01EKZ2BX1WE4H2WYP6MGTSF9", "gs:conn:01EKYZKSQRJHR9X9YDY0FVKV", "gs:uplink:01EKZ2BWVBYEP55D6WN27V683J", "ns:uplink:01EKZ2BWVC1WN4WTY01JDMRX", "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01EKZ2BWVCJGVAFHBSWSRCEX" ],
  "received_at" : "2020-10-06T13:42:14.332604900Z",
  "uplink_message" : {
    "session_key_id" : "AXT+JcAcFSPqwW+Wzxnh==",
    "f_port" : 2,
    "frm_payload" : "UEtUICMAAIBAZmbPQgAAgD+amaVBAACoQgAAj0LNzDRBMzMz==",
    "decoded_payload" : {
      "consumption" : 20.700000762939453,
      "device_type" : "J1939",
      "ignition" : 1,
      "motor_temp" : 84,
      "motor_water_temp" : 71.5,
      "oil_pressure" : 11.300000190734863,
      "remaining_fuel" : 103.69999694824219,
      "tire_pressure" : 2.799999952316284
    },
    "rx_metadata" : [ {
      "gateway_ids" : {
        "gateway_id" : "gtw1",
        "eui" : "30AEA4FFFEECB160"
      },
      "time" : "2020-10-06T13:42:02.506635Z",
      "timestamp" : 2894075524,
      "rssi" : -28,
      "channel_rssi" : -28,
      "snr" : 6,
      "uplink_token" : "ChIKEAoEZ3R3MRIIMK6k//7ssWAQhK2A5="
    } ],
    "settings" : {
      "data_rate" : {
        "lora" : {
          "bandwidth" : 125000,
          "spreading_factor" : 7
        }
      },
      "data_rate_index" : 5,
      "coding_rate" : "4/5",
      "frequency" : "868100000",
      "timestamp" : 2894075524,
      "time" : "2020-10-06T13:42:02.506635Z"
    },
    "received_at" : "2020-10-06T13:42:14.125004695Z"
  }
}**
```

I need to extrac data from the part: decoded_payload 

```
    "decoded_payload" : {
      "consumption" : 20.700000762939453,
      "device_type" : "J1939",
      "ignition" : 1,
      "motor_temp" : 84,
      "motor_water_temp" : 71.5,
      "oil_pressure" : 11.300000190734863,
      "remaining_fuel" : 103.69999694824219,
      "tire_pressure" : 2.799999952316284
    },
```

I don't know to configure mqtt.json to extract the data. I can subscribe well into **v3/machineman5/devices/+/up** but I can't extract data to my device in ThingsBoard. I have this configuration:
```
{
  "broker": {
    "name":"Default Local Broker",
    "host":"localhost",
    "port":18883,
    "security": {
      "type": "basic",
      "username": "machineman5",
      "password": "password"
    }
  },
  "mapping": [
    {
      "topicFilter": "v3/machineman5/devices/+/up",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=devices/)(.*?)(?=/)",
        "deviceTypeTopicExpression": "Machineman5",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "consumption" ,
            "value": "${compsumption}"
          },
          {
            "type": "string",
            "key": "device_type",
            "value": "${device_type}"
          },
          {
            "type": "double",
            "key": "ignition",
            "value": "${ignition}"
          },
          {
            "type": "double",
            "key": "motor_temp",
            "value": "${motor_temp}"
          },
          {
            "type": "double",
            "key": "motor_water_temp",
            "value": "${motor_water_temp}"
          },
          {
            "type": "double",
            "key": "oil_pressure",
	     "value": "${oil_pressure}"
          },
          {
            "type": "double",
            "key": "remaining_fuel",
            "value": "${remaining_fuel}"
          },
          {
            "type": "double",
            "key": "tire_pressure",
            "value": "${tire_pressure}"
          }
        ]
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
        {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
```

**Error traceback 
```
tb gateway service - 314 - Data from MQTT Broker Connector connector is invalid 
```

**Versions :**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway latest version

cristianhumelnicu, 2020-10-08:
Did figure out how to interface chirpstack with TB GATEWAY ?

zbeacon, 2020-10-22:
Hi @oscardavidtorres1994 ,

Thank you for your interest in ThingsBoard IoT gateway.

To parse data like this, you are able to use JsonPath Expressions, for example for cosumption value should be ${$.uplink_message.decoded_payload.consumption} , it will look like:

```
...

            "key": "consumption" ,
            "value": "${$.uplink_message.decoded_payload.consumption}"
          },
...

```

To check your JsonPath expression, you are able to use the following site - https://jsonpath.com/

zbeacon, 2020-10-22:
> Did figure out how to interface chirpstack with TB GATEWAY ?

Hi @cristianhumelnicu ,

As I know there is a MQTT integration in Chirpstack,  you may use it to connect the gateway using MQTT connector. Official documentation for MQTT integration in Chirpstack is [here](https://www.chirpstack.io/application-server/integrations/mqtt/).

cristianhumelnicu, 2020-10-22:
> Hi @cristianhumelnicu ,
>
> As I know there is a MQTT integration in Chirpstack, you may use it to connect the gateway using MQTT connector. Official documentation for MQTT integration in Chirpstack is [...]

Yes , i know about it , i try it and is working , but im much interested to connect TB gateway via MQTT to chirpstack

I mean how to configure mqtt from TB gatway for that. I know how to do it in old TB Gw ... but this version ... is over my knowledge.

Thx

zbeacon, 2020-10-22:
Please share your old configuration section mapping, I will try to help you to convert it to the current configuration.

cristianhumelnicu, 2020-10-22:
> Please share your old configuration section mapping, I will try to help you to convert it to the current configuration.

[TB-GW-LoraServer.txt](https://github.com/thingsboard/thingsboard-gateway/files/5422943/TB-GW-LoraServer.txt)

This is the old configuration.
Difference is topic, now is  application/123/device/+/event/up

Thak you very much for your help !

cristianhumelnicu, 2020-10-24:
> Please share your old configuration section mapping, I will try to help you to convert it to the current configuration.

Hi 
I put the configuration, can you help please ?

Thank You!

zbeacon, 2020-10-30:
Hi @cristianhumelnicu ,

Please try the following configuration:

```

{
  "broker": {
    "name":"Default Local Broker",
    "host":"localhost",
    "port":18883,
    "security": {
      "type": "basic",
      "username": "machineman5",
      "password": "password"
    }
  },
  "mapping": [
    {
      "topicFilter": "v3/machineman5/devices/+/up",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=devices/)(.*?)(?=/)",
        "deviceTypeTopicExpression": "Machineman5",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "consumption" ,
            "value": "${$.uplink_message.decoded_payload.consumption}"
          },
          {
            "type": "string",
            "key": "device_type",
            "value": "${$.uplink_message.decoded_payload.device_type}"
          },
          {
            "type": "double",
            "key": "ignition",
            "value": "${$.uplink_message.decoded_payload.ignition}"
          },
          {
            "type": "double",
            "key": "motor_temp",
            "value": "${$.uplink_message.decoded_payload.motor_temp}"
          },
          {
            "type": "double",
            "key": "motor_water_temp",
            "value": "${$.uplink_message.decoded_payload.motor_water_temp}"
          },
          {
            "type": "double",
            "key": "oil_pressure",
	     "value": "${$.uplink_message.decoded_payload.oil_pressure}"
          },
          {
            "type": "double",
            "key": "remaining_fuel",
            "value": "${$.uplink_message.decoded_payload.remaining_fuel}"
          },
          {
            "type": "double",
            "key": "tire_pressure",
            "value": "${$.uplink_message.decoded_payload.tire_pressure}"
          }
        ]
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
        {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}

```

cristianhumelnicu, 2020-10-30:
> Hi @cristianhumelnicu ,
>
> Please try the following configuration:
>
> ```
>
> {
>   "broker": {
>     "name":"Default Local Broker",
>     "host":"localhost",
>     "port":18883,
>     "security": {
>       "type": [...]

Hi Ilya,

My topic is application/+/device/+/event/up

Will be same mappig that u have sending to me?

Thank you very much!

zbeacon, 2020-10-30:
No, sorry, I have changed only mapping section, you should change `v3/machineman5/devices/+/up` to `application/+/device/+/event/up` . Also I recommend you change deviceNameTopicExpression to some static name for test purposes.

oscardavidtorres1994, 2020-10-30:
It works perfectly. Thanks a lot.

zbeacon, 2020-10-30:
I'm glad to hear this, please write us if you will have any questions/suggestions.

closed 2020-10-30 by zbeacon

Alwayssalt, 2021-05-24:
@zbeacon 
Hello
I used a configuration sample from ttn forum. But I still can not connect ttn mqtt broker to thingsboard mqtt connector.
Could you kindly help me?
My configuration file is as blew:
{
  "brokers": [
    {
      "host": "asia-se.thethings.network",
      "port": 1883,
      "ssl": false,
      "retryInterval": 3000,
      "credentials": {
        "type": "basic",
        "username": "applicationxxxxxx",
        "password": "ttn-account-v2.xxxxxxxxxxxxxxxxx"        
      },
      "mapping": [
        {
          "topicFilter": "+/devices/+/up",
          "converter": {
            "type": "json",
            "filterExpression": "",
            "deviceNameJsonExpression": "${$.dev_id}",
            "attributes": [
              {
                "type": "string",
                "key": "hardware_serial",
                "value": "${$.hardware_serial}"
              }
            ],
            "timeseries": [
              {
                "type": "long",
                "key": "humidity",
                "value": "${$.payload_fields.humidity}"
              },
              {
                "type": "double",
                "key": "temperature",
                "value": "${$.payload_fields.temperature}"
              }
            ]
          }
        }
      ]
    }
  ]
}