thingsboard/thingsboard-gateway #408: Pymodbus Word Order Configuration IoT Gateway [HELP]

adriortega, 2020-10-07:
Hi everyone

**Issue**

I am trying to read some data from a remote device via Modbus TCP. In order to configure my json files in config, I can change the "byteOrder", BIG or LITTLE. Nevertheless, it is impossible to change the "wordOrder" from the json files. I have look at pymodbus, and both parameters are included, but in the config files the "wordOrder" field is not allowed to be modified.
The data I am trying to read is codified in a Little endianness way, so when I have to send data which consists in more than 1 word, I have problems with the format. 

So I would like to know how can I modified the modbus TCP or the IoT gateway configuration so as to, being able to read correctly my data. Which configuration files I have to change, because if it is included in pymodbus, I suppose that the modification is possible.

**Configuration** (Attach your configuration file)
**In pymodbus, it runs ok, and is possible to change the configuration:**
# The Word order is applicable only for 32 and 64 bit values
    # Lets say we need to write a value 0x12345678 to a 32 bit register
    # The following combinations could be used to write the register   # +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ #
    # Word Order - Big                      Byte Order - Big
    # word1 =0x1234 word2 = 0x5678
 
    # Word Order - Big                      Byte Order - Little
    # word1 =0x3412 word2 = 0x7856
 
    # Word Order - Little                   Byte Order - Big
    # word1 = 0x5678 word2 = 0x1234
 
    # Word Order - Little                   Byte Order - Little
    # word1 =0x7856 word2 = 0x3412
    # +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ #

** In config files, JSON, I can only configure the byteOrder field:**
"server": [
    {
      "type": "tcp",
      "host": "192.168.1.36",
      "port": 502,
      "timeout": 100,
      "method": "socket",
      "byteOrder": "BIG",
      "devices": [
        {
          "unitId": 6,
          "deviceName": "TCU_002_Cordovilla",
          "attributesPollPeriod": 30000,
          "timeseriesPollPeriod": 30000,
          "sendDataOnlyOnChange": false,
          "timeseries": [
            {
              "tag": "ProductId",
              "type": "16uint",
              "functionCode": 4,
              "objectsCount": 1,
              "address": 30000
            }   ]    }  ]  },

**Versions (please complete the following information):**
 - OS: Windows 7
 - Thingsboard IoT Gateway version 2.5.0
 - Python version 3.7

I hope that I have explained it clearly, and that there is an esay solution.

Regards

zbeacon, 2020-10-09:
Hi @adriortega ,

Thank you for your feedback. I have write support for this configuration, parameter with name "wordOrder". The version with these improvements is available in master branch and in develop/2.4-python.

Please try to set up it and use. Also please let me know about the results.

adriortega, 2020-10-09:
Thank you very mach @zbeacon . I will see if this modification can be enough for my purposes.

Regards.

closed 2020-10-30 by zbeacon