thingsboard/thingsboard-gateway #412: [HELP]

NullPJ, 2020-10-09:
Hi  everyone
**Describe the issue**
I use Modbus tcp Connector , but it happend error    

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[Modbus  Connector]

**Error traceback (If it was raised):**

""2020-10-09 10:32:40" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 72 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2020-10-09 10:32:40" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 72 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2020-10-09 10:32:40" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 72 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2020-10-09 10:32:40" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 72 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2020-10-09 10:32:40" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 72 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None

**Versions (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.5.1]
 - Python version[ 3.7]

zbeacon, 2020-10-09:
Hi @NullPJ 

It looks like your Modbus server closed connection, unfortunately, you have not provided configuration so we cannot reproduce the issue.

closed 2020-10-30 by zbeacon

yvo-pfenninger, 2020-12-23:
@NullPJ:
have you been able to solve this issue?

@zbeacon 
It looks that I have the same issue:

""2020-12-23 10:43:17" - DEBUG - [bytes_modbus_downlink_converter.py] - bytes_modbus_downlink_converter - 95 - [11547]"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 301 - With result Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 371 - ModbusIOException(ConnectionResetError(104, 'Connection reset by peer'), 6)"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 237 - Modbus trying connect to KEBA_PRiOT005"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 239 - Modbus connected."
""2020-12-23 10:43:17" - DEBUG - [bytes_modbus_downlink_converter.py] - bytes_modbus_downlink_converter - 95 - [0]"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 301 - With result Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 371 - ModbusIOException(ConnectionResetError(104, 'Connection reset by peer'), 6)"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 237 - Modbus trying connect to KEBA_PRiOT005"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 239 - Modbus connected."

Versions (please complete the following information):

Debian GNU/Linux 10 (buster)
Kernel: 5.4.24
docker/stable 1.5-2 all
Thingsboard IoT Gateway version 2.5.5.2
python3/stable 3.7.3-1 armhf

my current modbus.json config file

{
  "server": {
    "name": "Modbus Default Server",
    "type": "tcp",
    "host": "192.168.11.152",
    "port": 502,
    "timeout": 35,
    "method": "socket",
    "byteOrder": "BIG",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "TCP_Slave005",
        "deviceType": "Type005",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "cable_state",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1004,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "firmware_code",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1013,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "charging_state",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1000,
[... 184 lines cut ...]
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "unlock_plug",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 5012,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "enable_user",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 5014,
            "deviceName": "TCP_Slave005"
          }
        ]
      }
    ]
  }
}