thingsboard/thingsboard-gateway #418: [BUG] mqtt_connector.py

CristoJV, 2020-10-13:
**Describe the bug**
When using AttributeRequests configuration parameter it fails silently.
This is the specific part of the mqtt connector's configuration json (mqtt.json)

```
"attributeRequests": [
    {
      "topicFilter": "devices/+/light/request",
      "deviceNameTopicExpression": "(?<=devices\/)(.*?)(?=\/light)",
      "attributeNameJsonExpression": "${attribute}",
      "topicExpression": "devices/${deviceName}/light/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":${attributeValue}}"
    }
  ],
```
It is important to highlight that the `attributeValue` is a NUMBER and that it what causes the failure.

I have debugged the running application and found that the issue is inside the ```notify_attribute``` method.

```
def notify_attribute(self, incoming_data, attribute_name, topic_expression, value_expression):
    if incoming_data.get("device") is None or incoming_data.get("value") is None:
           return

        device_name = incoming_data.get("device")
        attribute_value = incoming_data.get("value")
        topic = topic_expression \
            .replace("${deviceName}", device_name) \
            .replace("${attributeKey}", attribute_name)
        
        !!!HERE ITS WHERE IT FAILS SILENTLY!!!"
        data = value_expression.replace("${attributeKey}", attribute_name) \
            .replace("${attributeValue}", attribute_value)
        self._client.publish(topic, data).wait_for_publish()
```

The issue is that the replace method does not expect a number but a string when it tries to replace the ```${attributeValue}``` expression for the ```attribute_value``` value.

The solution is a simple string casting of two variables (at least the attribute_value variable):

- the device_name variable 
- the attribute_value variable.

This can be done with minor to none repercussions at the start of this method.

```
def notify_attribute(self, incoming_data, attribute_name, topic_expression, value_expression):
    if incoming_data.get("device") is None or incoming_data.get("value") is None:
           return

        <----- Modifications begin ----->
        device_name = str(incoming_data.get("device"))
        attribute_value = str(incoming_data.get("value"))
        <----- Modifications end ----->

        topic = topic_expression \
            .replace("${deviceName}", device_name) \
            .replace("${attributeKey}", attribute_name)
        data = value_expression.replace("${attributeKey}", attribute_name) \
            .replace("${attributeValue}", attribute_value)
        self._client.publish(topic, data).wait_for_publish()
```

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
There is no error. It fails silently.

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

As I'm fairly new in Github I would like to contribute to solve this issue if you find that it can be solved as I'm stating here.

Thank you!

zbeacon, 2020-10-20:
Hi @CristoJV ,

Thank you for your investigations, I checked it and discovered more than one issue like this in the MQTT connector. Anyway, I have fixed it and you are able to use the version without this issue. Please install the version of the gateway from the master branch and check it. Please write us if you find any issues in the gateway. Also the fix will be added to the next release of the gateway.

zbeacon, 2020-10-22:
Hi @CristoJV ,

Fix is available in the release 2.5.2.

Thank one more time for your feedback, if you discover any issue, please write us.

closed 2020-10-22 by zbeacon