thingsboard/thingsboard-gateway #421: [BUG] RTU Connection is not established.

jocatelo, 2020-10-23:
**Describe the bug**
RTU Connection is not established.

**Connector name (If bug in the some connector):**
Modbus Connector.

**Error traceback (If available):**

at the modbus_connector.py nearly 240 line, It intercept rtu connection.
```
log.debug("Modbus trying connect to %s:%r", self.__current_master.host,
                 self.__current_master.port)
```

**Versions (please complete the following information):**
 - OS: Raspberry
 - Thingsboard IoT Gateway version [e.g. 2.5.2]
 - Python 3

zbeacon, 2020-10-27:
Hi @jocatelo 

Thank you for your feedback. I have done fix for this issue, please try to install the gateway from the master branch of the current repository.

jocatelo, 2020-10-28:
After executing thingsboard-gateway, it reports error:
```
Exception in thread Modbus Connector:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.2-py3.7.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 72, in run
    self.__connect_to_current_master()
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.2-py3.7.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 233, in __connect_to_current_master
    device_address = str(self.__current_master.host) + ":" + str(self.__current_master.port) if self.__devices[device]["config"]["type"] != 'serial' else self.__current_master.port
KeyError: None
```

zbeacon, 2020-10-28:
Thank you for information.
Please try the current version from the master branch.

closed 2020-12-02 by zbeacon