thingsboard/thingsboard-gateway #422: [HELP] Disconnected client: <paho.mqtt.client.Client object at 0x7f59601ce9d0>, user data: None, result code: 1

venushong667, 2020-10-28:
**Describe the issue**
I'm currently implementing tb-gateway on python, trying to receive the telemetry data from client/server.
I faced the issue above when I'm connecting devices through gateway either mqtt or opcua 
Everything works fine, the telemetry data and events, logs on thingsboard are all good, but the log on both gateway and thingsboard server keep showing as the images as long as the connection is up
From what I see it's connected to server getting one time telemetry data and disconnect after that, repeating for every data receiving event
Is it purposely building like this one client just to collect one time data? if not, I would like to know why is it happens?

Thingsboard side:
[image]
*The blue cover username is actually the access token of device

Gateway side:
[image]

Below is the configuration that I used, it's basically the default setting with docker rabbitmq.

**Configuration** (Attach your configuration file)
mqtt.json
```
{
  "broker": {
    "name":"Default Local Broker",
    "host":"rabbit",
    "port":1883,
    "security": {
      "type": "basic",
      "username": "username",
      "password": "password"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "${key1}",
            "value": "${value1}"
          },
          {
            "type": "double",
            "key": "${key2}",
            "value": "${value2}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}

```

tb-gateway.yaml
```
connectors:
- configuration: thingsboard_gateway/config/mqtt.json
  name: mqtt Connector
  type: mqtt
storage:
  max_records_count: 100000
  read_records_count: 100
  type: memory
thingsboard:
  host: localhost
  port: 1883
  security:
    accessToken: ACCESSTOKEN
```

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Versions (please complete the following information):**
 - OS: Windows 10
 - Thingsboard IoT Gateway version 2.5.1
 - Python version 3.7

zbeacon, 2020-10-28:
Hi @venushong667,

Thank you for your feedback, usually the issues like this appear when several gateways are running at the same time. As we can see from your logs, one session (Idenfied by UUID in start of log record) is closed and another one opened. Please check one more time that the only one gateway is running, because ThingsBoard doesn't support more than one device per token.

venushong667, 2020-10-28:
> Hi @venushong667,
>
> Thank you for your feedback, usually the issues like this appear when several gateways are running at the same time. As we can see from your logs, one session (Idenfied by UUID in [...]

First of all, thanks for the reply.
That's my first thought too, but I actually tried out with a new device, even changing the thingsboard server to demo.thingsboard, with clean python file that hold the tb-gateway package, I got all the same result, repeating client connections

So, I actually tried out the docker run tb-gateway too, is it the correct logging if everything works fine? In the image, I'm connecting with demo server but the connections looks like keep getting new connections
While I'm connecting to my local tb server, it has the same log and my local tb server logging show the same thing as above, client connect and disconnect again and again
[image]

zbeacon, 2020-10-28:
Could you try the version from the master branch, instead of docker?

venushong667, 2020-10-29:
It looks like having the same result, I wonder is it my client issue, since I'm actually testing with nodejs mqtt client
Another question, I know that gateway is only available for one token (one device), so how about if I use multiprocessing or create workers for gateway with every single token? Multiple gateway with different tokens connecting to the same thingsboard server should be fine right?
Or is there any better options for connecting more than one device at the same time?

zbeacon, 2020-10-29:
Unfortunately, there is no way to use multiprocessing for communication between ThingsBoard and devices. But you can actually use gateways with different access tokens and the same devices in configurations.

venushong667, 2020-10-29:
What do you think about if the gateway working with docker workers for every single device (token), it works fine actually, but I couldn't solve the issue above, function is works but the logging looks weird and I'm afraid of the connect/disconnect traffic issues
Back to the question, I tried out with docker, master branch or python packages, all got the same result that when I publish a single message, it works the same Client Disconnected at first, processing message with no username, processing message with username: ${accessToken}, and client connected 
The behavior is like it client disconnect to process the message, after the message is processed and display on Thingsboard UI, it show client connected again.

zbeacon, 2020-10-29:
Please make sure that every docker image use his own configuration. I didn't see the issues like this in other cases. If you don't find the cause of the issue please email me ibarkov@thingsboard.io , I will connect and try to find the cause of the issue.

venushong667, 2020-10-30:
I'm not sure whether it will be the reason or not, I found that for mqtt message if there is second CONNECT packet sent to the broker, it will ignored and disconnect the client.
In my case, I actually got this message below when I connect my gateway to TB server, it treat gateway as Client and sent a connect packet, is it expected behavior?
[image]

Then I started a nodejs mqtt client connect to the same broker and publish message getting the log as below, the client(gateway) disconnected at first and reconnected back after that
[image]

Will it possible that it caused by the duplicated of CONNECT packet?

zbeacon, 2020-10-30:
Yes, it is expected behavior. As we can see from screenshots, there are different session identifiers in logging record. The disconnecting caused by using the same credentials for connection and ThingsBoard disconnects old session and opens a new one. It is architects decision and I cannot change it in ThingsBoard. To avoid this situation you should not use the same accessToken for different clients at the same time.

closed 2020-11-23 by zbeacon