thingsboard/thingsboard-gateway #423: modbus rtu  connnect also bug report [BUG]

lyly0906, 2020-10-28:
**Describe the bug**

Exception in thread Modbus:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/site-packages/thingsboard_gateway-2.5.2-py3.6.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 72, in run
    self.__connect_to_current_master()
  File "/usr/local/lib/python3.6/site-packages/thingsboard_gateway-2.5.2-py3.6.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 233, in __connect_to_current_master
    device_address = str(self.__current_master.host) + ":" + str(self.__current_master.port) if self.__devices[device]["config"]["type"] != 'serial' else self.__current_master.port
KeyError: None

**Connector name (If bug in the some connector):**
[e.g. modbus RTU Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

zbeacon, 2020-10-28:
Hi @lyly0906 ,

Thank you for your feedback, could you try the current version from the master branch, it has to work fine.

lyly0906, 2020-10-28:
yes,I will git clone the newest version from the master branch.  it's also bug report.       
[image]

zbeacon, 2020-10-28:
Ok, I removed getting the port from the connector, please try again. If it doesn't work please attach your configuration for Modbus connector.

lyly0906, 2020-10-28:
[image]

My  Modbus RTU configuration is :  

{
  "name": "Modbus",
  "server": {
    "baudrate": 9600,
    "byteOrder": "BIG",
    "devices": [
      {
        "attributes": [],
        "attributesPollPeriod": 3600000,
        "deviceName": "Air Humi and Temp",
        "deviceType": "default",
        "rpc": [],
        "sendDataOnlyOnChange": false,
        "timeseries": [
          {
            "address": 0,
            "byteOrder": "BIG",
            "deviceName": "Air Humi and Temp",
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "humidity",
            "type": "long"
          },
          {
            "address": 1,
            "byteOrder": "BIG",
            "deviceName": "Air Humi and Temp",
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "temperature",
            "type": "long"
          }
        ],
        "timeseriesPollPeriod": 3000,
        "unitId": 1
      }
    ],
    "method": "rtu",
    "name": "Modbus Default Server",
    "port": "/dev/ttyUSB0",
    "timeout": 35,
    "type": "serial"
  }
}

zbeacon, 2020-10-28:
I apologize, I was looking for in the wrong place, please try now.

lyly0906, 2020-10-28:
Thanks。  Now the modbus rtu  is ok。
[image]

commit 0fcbb68c (2020-10-28): Fixed #423
closed 2020-10-28 by zbeacon