thingsboard/thingsboard-gateway #426: How to make more efficiency when poll device with bacnet protocol

spemed, 2020-11-03:
Hello
I met some problem when poll our devices with bacnet protocol.I have two hundred of devices which configured on bacnet.json,and each device has more than 5 attribute。It means that there are 200 * 5 = 1000 udp datagram waiting for sending to my devices. And it need at last 75s to consume all message in once poll between the first udp datagram sent and the last udp datagram received. It leads to a lower timeliness of our program. I want to know whether there is a batch api to packet the query of multiple attributes in a single UDP datagram for decreasing the number of UDP datagram.

lbkk523, 2020-11-10:
@spemed Hello! I encountered a problem when connecting to a bacnet device. Currently, only one bacnet device can be connected to thingsboard. I want to see how your bacnet.json is configured to connect multiple bacnet devices to thingsboard. Learn from your configuration method. can you help me?​

spemed, 2020-11-10:
hello !
My bacnet.json content like this
------
{
  "devices": [
    {
      "address": "xx.xx.xx.xx",
      "attributes": [
        {
          "key": "AF-A",
          "objectId": "binaryInput:3000239",
          "propertyId": "presentValue",
          "type": "boolean"
        }
      ],
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "keepaliveRound": 3,
      "pollPeriod": 2730,
      "serverSideRpc": [
        {
          "method": "set_CO2-LSP",
          "objectId": "analogValue:3000288",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "writeProperty"
        },
        {
          "method": "get_CO2-LSP",
          "objectId": "analogValue:3000288",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "readProperty"
        },
        {
          "method": "set_OAD-MIN",
          "objectId": "analogValue:3000295",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "writeProperty"
        },
        {
          "method": "get_OAD-MIN",
          "objectId": "analogValue:3000295",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "readProperty"
        },
        {
          "method": "set_RA-HSP",
          "objectId": "analogValue:3000286",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "writeProperty"
        },
        {
          "method": "get_RA-HSP",
          "objectId": "analogValue:3000286",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "readProperty"
        },
        {
          "method": "set_RA-TSP",
          "objectId": "analogValue:3000289",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "writeProperty"
        },
        {
[... 211 lines cut ...]
        },
        {
          "method": "get_WS-EX",
          "objectId": "multiStateValue:3000293",
          "propertyId": "presentValue",
          "requestTimeout": 10000,
          "requestType": "readProperty"
        }
      ]
    }
  ],
  "general": {
    "address": "0.0.0.0",
    "maxApduLengthAccepted": 1024,
    "objectIdentifier": 599,
    "objectName": "TB_gateway",
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  }
}
------
Type of attribute "devices"  is array.if you want to configure multi devices you should append device item.
And make sure that device ip is reachable.
You can use the tools bacpypes to validate if the device is reachable
[https://bacpypes.readthedocs.io/en/latest/gettingstarted/gettingstarted001.html](url)

lbkk523, 2020-11-10:
@spemed thank you very much，let me have a try

closed 2021-11-15 by zbeacon