thingsboard/thingsboard-gateway #427: wordOrder field does not produce any effect in my RPC Call Requests [HELP]

adriortega, 2020-11-04:
**Describe the issue**

Hi, I have been making **RPC requests in modbus tcp**, in order to change the values of some different type of variables in devices connected to thingsboard gateway. With 16uint or 16int data types, there is not any problem writing in devices' memory, but when I was trying to do so in 32float or 32uint variables, the changes are not being done well.
I have been trying to follow the data path, so I can see the exact moment when the failure is produced, and I think that I may find it.
Comparing my sent values to the ones are written in memory and shown in the device screens in binary and hexadecimal format, there is an evident mistake, for example with the RPC  _“setHome_a1_rad”_ : 
### 
**Ex 1**: When I sent 4.74º : **0.082728** radians, in floating point format:  0b0011 1101 1010 1001 0110 1101 1001 1110 ; **0x3DA9 6D9E** . On the contrary, in the device is written the following data:  0011 1101 1011 0010 0110 1101 1001 1110: **0x3DB2 6D9E**, which means **0.0871231** radians or 4.99179º.

**Ex 2**: When I sent 7.6º: **0.132645** radians, in floating point format:  0b0011 1110 000 0111 1101 0100 0001 1001: **0x3E07 D819** . Onthe contrary, in the device is written the following data:  0b0011 1101 1011 0010 1101 0100 0001 1001: **0x3DB2 D819**, which means **0.08731864** radians or 5.00299º.
### 
As you can see, in both examples, every byte is being sent and written correctly except the least significant byte of the most significant word. 
Thats why I think that what is happening is that I am not allowed to use the wordOrder field, so the most siginificant word remains always the same, in its predeterminte value.

**My question is : What can I do to be able to write the 32 float data properly??** 
### 
Because I think that the most significant word of the 32float is never changing.

**Configuration** (Attach your configuration file)
In the config file(json): 

"rpc": {

        "setHome_a1_rad": {
            "type": "32float",
            "value": 0,
            "functionCode": 16,
            "address": 41042,
            "byteOrder": "BIG",
            "tag": "variable",
            "objectsCount": 2,
            "wordOrder": "LITTLE"
          }
   }

**Connector name (If you need help with some connector/converter):**
    Modbus TCP Connector

**Versions (please complete the following information):**
 - OS: [Windows 7]
 - Thingsboard IoT Gateway version  2.5.0.
 - Python version 3.7.

zbeacon, 2020-11-05:
Hi @adriortega ,

Thank you for your feedback, it is strange, because it should use wordOrder property.
Anyway, could you try to use the version from the master branch and update library for modbus?
To update the library for modbus, you can use the following command: `pip3 install pymodbus --upgrade`
Also, please try the latest version of the gateway, wordOrder paraameter for Modbus connector was added in 2.5.1 and improved in 2.5.2.

adriortega, 2020-11-09:
Hi @zbeacon 

Thank you very much for your help. I have just upgrade the new version of pymodbus and thingsboard gateway and the problem remains the same.  The upper word of the 32float registers doesn´t change when I tried to send new data, although it was supposed to. 
Previously, I had similar problems with the input readings, wordorder was not included in the gateway configuration, may it be the reason?

thanks and regards.

zbeacon, 2020-11-16:
Hi @adriortega ,

Did you install the gateway from sources?

adriortega, 2020-11-16:
Hi again @zbeacon ,

I upgrade the gateway and pymodbus versions with the pip commands.

Is it correct??

Regards

Adrián.

zbeacon, 2020-11-16:
Could you try to install the version from sources or we will release new version of the gateway at the end of week.

adriortega, 2020-11-23:
Hi @zbeacon .

Last week I installed the gateway version from sources and the problem still remains. Could you tell me if you have upgrade it recently, or if you will do so in a close future? 

Thank you very much

adriortega, 2020-12-03:
Hi again @zbeacon.

As I said you previously, I had a problem with the RPC calls in my gateway. In order to solve it, I was looking at the modbus downlink converter library, trying to understand the possible origin of the problem.  I have modified some lines of this file, which I think that they didn´t allow my gateway to perform its duties correctly, including the gateway latest release, 2.5.4.

Here, I explain you the changes I have committed, with the purpose of analysing it, and change whatever you will think it may be neccesary to include in the following version:
 **First Change:  Line 55:**, the variable_size must be multiplied by 16 instead of 8:
     
**Current Code**: variable_size = config.get(“objectsCount”, config.get(“registersCount”,  config.get(“registerCount”, 1))) * 8

**Modified Code**: variable_size = config.get(“objectsCount”, config.get(“registersCount”,  config.get(“registerCount”, 1))) * 16

**Second Change: Line 99** : When the object is a 32float, the length (len) is 2, and currently it asures that it should be 8, 16, 32 or 64, and that doesn´t make any sense. My new proposal would be the next:

**Current Code:**   if isinstance(builder, list) and len(builder) not in (8, 16, 32, 64):
                builder = builder[0]
            return builder

**Modified Code**:    if variable_size<=16:
                if isinstance(builder, list) and len(builder) not in (8, 16, 32, 64):
                   builder = builder[0]
            else:
                if isinstance(builder, list) and len(builder) not in (2, 4):
                    builder = builder[0]
          return builder

Those are the changes I committed in my host, I have achiveved very good results, so I recommend you to check the current code and think about possible changes or upgrades.

Regards.

zbeacon, 2020-12-03:
Hi @adriortega ,

Thank you for your investigations and fixes, could you make a pull request to this repository?

adriortega, 2020-12-03:
Of course @zbeacon , tomorrow I will make the request with my new version. I hope that it will be successful for all of us.

closed 2021-03-01 by zbeacon