thingsboard/thingsboard-gateway #428: [HELP] Modbus Connector | AttributeError: 'ModbusSerialClient' object has no attribute 'host'

smit-github, 2020-11-05:
**Describe the issue**
Create description about your issue, and your actions to solve it.
I have installed the thingsboard IoT gateway in rPi3 ,I am trying to connect the modbus slave using modbus connector of the IoT gateway thingsboard. and got below error i have attached usb to uart in rPi3 .

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**
{
  "server": {
    "name": "Modbus Default Server",
    "type": "serial",
    "method": "rtu",
    "port": "/dev/ttyUSB0",
    "baudrate": 19200,
    "timeout": 35,
    "devices": [
      {
        "unitId": 1,
        "deviceName": "Temp Sensor",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "test",
            "type": "long",
            "functionCode": 4,
            "registerCount": 1,
            "address": 4
          }
        ]
      }
    ]
  }
}

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
Modbus Serial Connector 

**Error traceback (If it was raised):**
```
Exception in thread Modbus Connector:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 72, in run
    self.__connect_to_current_master()
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 233, in __connect_to_current_master
    log.debug("Modbus trying connect to %s:%r", self.__current_master.host,
AttributeError: 'ModbusSerialClient' object has no attribute 'host'

```

**Versions (please complete the following information):**
 - OS: raspbian {raspberry pi3B+}
 - Thingsboard IoT Gateway version 2.2.4
 - Python version .7

zbeacon, 2020-11-05:
Hi @smit-github ,

Please try to use the version from master branch. This os known issue and it is already resolved in master branch.

smit-github, 2020-11-05:
which version it has ?

zbeacon, 2020-11-05:
It is version 2.5.2, so ro install it, you should run the following
commands
sudo pip3 uninstall thingsboard-gateway

And

pip3 uninstall thingsboard-gateway

It will remove the old version from your system and then open repository
folder and run

sudo python3 setup.py install

smit-github, 2020-11-05:
I have done the above steps now i didn't able to find the how to configure using config file with above installation using source option as you above mentioned!

zbeacon, 2020-11-06:
How did you run the gateway - as a daemon or by command from the "installation from sources" guide?

smit-github, 2020-11-06:
by command that is given in installation from sources as suggested in "installation from sources"

zbeacon, 2020-11-06:
Ok, in this case the gateway takes configuration from config folder in thingsboard_gateway folder in cloned repository folder.

smit-github, 2020-11-06:
> I have done the above steps now i didn't able to find the how to configure using config file with above installation using source option as you above mentioned! as modbus error solved by installing as per your method but got below error
> 
> ```
> 2020-11-05 18:23:49' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:50' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:52' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:53' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:54' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:56' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:57' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:58' - DEBUG - tb_client - 125 - connecting to ThingsBoard'
> ''2020-11-05 18:23:59' - DEBUG - tb_client - 83 - TB client <paho.mqtt.client.Client object at 0x75b11290> connected to ThingsBoard'
> ''2020-11-05 18:23:59' - ERROR - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised'
> ''2020-11-05 18:23:59' - DEBUG - tb_device_mqtt - 125 - Disconnected client: <paho.mqtt.client.Client object at 0x75b11290>, user data: None, result code: 5
> ```
> 
> `

what about the errors above mentioned i have got that configuration files and changes the token of as per thingsboard but now it is giving above errors

smit-github, 2020-11-06:
thingsboard-gateway not conneted to the thingsboard

```
""2020-11-06 16:47:25" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised"
""2020-11-06 16:47:36" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised"
""2020-11-06 16:47:47" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised"
""2020-11-06 16:47:57" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised"
""2020-11-06 16:48:08" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 148 - connection FAIL with error 5 not authorised"

```
Please HELP