thingsboard/thingsboard-gateway #430: [BUG] Import BACnetConnector, wrong path (ThingsBoard IoT Gateway using Docker)

ingempo, 2020-11-18:
**Describe the bug**
Error log in tb_utility module when run on Docker. 

**Connector name (If bug in the some connector):**
[tb_utility]

**Error traceback (If available):**
tb-gateway    | ""2020-11-18 04:43:57" - INFO - [tb_gateway_service.py] - tb_gateway_service - 72 - Gateway starting..."
tb-gateway    | ""2020-11-18 04:43:57" - INFO - [tb_gateway_service.py] - tb_gateway_service - 77 - ThingsBoard IoT gateway version: 2.5.2"
tb-gateway    | ""2020-11-18 04:43:57" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 1"
tb-gateway    | BACnet library not found - installing...
tb-gateway    | ""2020-11-18 04:43:57" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
tb-gateway    | Collecting bacpypes>=0.18.0
tb-gateway    |   Downloading bacpypes-0.18.3-py3-none-any.whl (184 kB)
tb-gateway    | Installing collected packages: bacpypes
tb-gateway    | Successfully installed bacpypes-0.18.3
tb-gateway    | ""2020-11-18 04:43:59" - ERROR - [tb_utility.py] - tb_utility - 110 - Import BACnetConnector failed, path /var/lib/thingsboard_gateway/extensions/bacnet doesn't exist"
tb-gateway    | ""2020-11-18 04:43:59" - WARNING - [tb_gateway_service.py] - tb_gateway_service - 311 - Connector implementation not found for BACnet Connector"
tb-gateway    | ""2020-11-18 04:43:59" - INFO - [tb_gateway_service.py] - tb_gateway_service - 135 - Gateway started."

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04 + Docker
 - Thingsboard IoT Gateway version 2.5.2
 - Python version 3.7

zbeacon, 2020-11-18:
Hi @ingempo ,

Thank you for your message, we are able to see the first run of the connector in logs - the gateway installs required library, you should restart the gateway using command `docker restart tb_gateway`. It will reload and works correct. Please try and let us know about results.

ingempo, 2020-11-30:
it's OK. New Bug comming.

closed 2020-11-30 by ingempo