thingsboard/thingsboard-gateway #431: [HELP] clide-Side RPC via external mqtt broker (mqtt) and thingsboard gateway

lillianphyoe, 2020-11-23:
**Describe the issue**
Our requirement is to subscribe topic with basic mqtt from external mqtt broker. I tried with thingsboard gateway and mosquitto mqtt broker. I can sent attributes and telemetry data to thingsboard using mosquitto_pub. Please let me know how can I retrive data using mosquitto_sub.

**Configuration** (Attach your configuration file)   
`thingsboard:
  host: tb_mqtt_transport_host
  port: tb_mqtt_transport_port
  security:
    accessToken: #YOUR_GW_ACCESS_TOKEN
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

  -
    name: BLE Connector
    type: ble
    configuration: ble.json
`
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
{
  "broker": {
    "name":"Default Local Broker",
    "host":"127.0.0.1",
    "port":1883,
    "security": {
      "type": "basic",
      "username": "user",
      "password": "password"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
[e.g. MQTT Connector]

**Versions (please complete the following information):**
 - OS: [e.g. Centos8]
 - Thingsboard IoT Gateway version [e.g. latest]
 - Python version[e.g. 3.7]

zbeacon, 2020-11-23:
Hi @lillianphyoe,

Thank you for your interest in ThingsBoard IoT gateway.
The gateway doesn't contain inside MQTT broker, but you are able to subscribe to topic on external broker. e.g. sensor/# , then you will be able to see topic and message that was sent to the external broker. RPC request from ThingsBoard sends to topic, generated using "requestTopicExpression" in the serverSideRpc configuration section.

lillianphyoe, 2020-11-23:
Do you mean I have to subscribe data from external broker with topic /sensor/SN-001/data that I sent.I want to know when I subscribe topic from external mqtt, gateway help to fetch data from thingsboard attribute value? I tried this for couple of weeks ago. If my question is stupid, I am sorry. Please let me know also subscription way also. mosquitto_sub -h my_host -p my_port -t {{ I don't know which topic to subscribe }}.

zbeacon, 2020-11-23:
Yes, you are correct, you should subscribe to /sensor/SN-001/data on external broker, if you wanna receive attribute "data" update request for device "SN-001". Mosquitto command is mosquitto_sub -h my_host -p my_port -t /sensor/SN-001/data
Also please make sure that device is active on ThingsBoard when you update the shared attribute. To check it you should check is server device attribute "active" equal to true.

lillianphyoe, 2020-11-23:
[image]
Can you please help, what is wrong my subscription sincerely?

I am trying to subscribe RPM value client attribute.
[image]

status is active as you said. :-(
[image]

zbeacon, 2020-11-24:
Hi @lillianphyoe ,

thingsBoard sends attribute update only when shared attribute was updated, if you need to get client, you are able to configure rule chain to save attribute to the SHARED_SCOPE instead of CLIENT_SCOPE.

lillianphyoe, 2020-11-24:
Thank you so much for your patience.Although I can subscribe share attribute from direct tb_mqtt_transport, I cannot subscribe share attribute from external broker. 
[image]

[image]

zbeacon, 2020-12-07:
Hi @lillianphyoe ,

The gateway subscribes to described topic (v1/gateway/attributes), transform data and send it to topic from configuration. ThingsBoard doesn't allow to subscribe and get shared attribute updates twice. Also you should send RPC to device instead of gateway.

lillianphyoe, 2020-12-07:
Thank you so much for your help. I will follow RPC for it.

closed 2020-12-07 by lillianphyoe