thingsboard/thingsboard-gateway #432: 'NoneType' object has no attribute 'upper'

io-r-ahmed, 2020-11-29:
**Describe the issue**
Create description about your issue, and your actions to solve it.
I am having an issue integrating modbus serial with thingsboard gateway, where the device is created on thingsboard and modbus is connected but no attributes is being created providing at the log this error "'NoneType' object has no attribute 'upper'"
At first I was working with thingsboard gateway as a package the one at this link https://thingsboard.io/docs/iot-gateway/install/deb-installation/ then after following different issues I installed the latest version from source but the same error is still here, please help!

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
Modbus Serial

**Error traceback (If it was raised):**
""2020-11-29 10:36:56" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-11-29 10:36:56" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 110"
""2020-11-29 10:36:58" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 84 - 'NoneType' object has no attribute 'upper'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 48, in convert
    word_endian_order = Endian.Little if word_order.upper() == "LITTLE" else Endian.Big
AttributeError: 'NoneType' object has no attribute 'upper'
""2020-11-29 10:36:58" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-11-29 10:36:58" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 111"
""2020-11-29 10:37:01" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-11-29 10:37:01" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 112"

**Versions (please complete the following information):**
 - OS:  armv7l GNU/Linux
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

zbeacon, 2020-12-01:
Hi @io-r-ahmed ,

Thank you for your message it is a bug, I will fix it in new release. As a workaround you are able to add parameters `"wordOrder":"LITTLE"` and `"byteOrder":"BIG"` into timeseries object configuration.

io-r-ahmed, 2020-12-01:
Hello, 
Thanks a lot for your reply.
I have done so where my final config file is:
{

  "server": {

    "name": "Modbus Default Server",

    "type": "serial",

    "method": "rtu",

    "port": "/dev/ttyUSB0",

    "baudrate": 9600,

    "timeout": 35,

    "devices": [

      {

        "unitId": 1,

        "deviceName": "Serial Machine",

        "attributesPollPeriod": 5000,

        "timeseriesPollPeriod": 5000,

        "sendDataOnlyOnChange": true,

        "attributes": [

          {

            "wordorder": "LITTLE",

            "byteOrder": "BIG",

            "tag": "test",

            "type": "long",

            "functionCode": 4,

            "registerCount": 1,

            "address": 4

          }

        ]

      }

    ]

  }

}

 

But the same error is still not solved.

Best Regards.

io-r-ahmed, 2020-12-01:
And Kindly check the attached log file on running thingsboard for more details.

Best Regards.

 

 

 

 

""2020-12-01 13:06:58" - INFO - [tb_gateway_service.py] - tb_gateway_service - 72 - Gateway starting..."
""2020-12-01 13:06:58" - DEBUG - [tb_updater.py] - tb_updater - 69 - Checking for new version"
""2020-12-01 13:06:58" - INFO - [tb_gateway_service.py] - tb_gateway_service - 77 - ThingsBoard IoT gateway version: 2.5.4"
""2020-12-01 13:06:58" - DEBUG - [tb_client.py] - tb_client - 125 - connecting to ThingsBoard"
""2020-12-01 13:06:58" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 301 - Subscribed to * with id 1"
""2020-12-01 13:06:58" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 1"
""2020-12-01 13:06:58" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log service"
""2020-12-01 13:06:58" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log extension"
""2020-12-01 13:06:58" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log converter"
""2020-12-01 13:06:58" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log connector"
""2020-12-01 13:06:58" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log tb_connection"
""2020-12-01 13:06:58" - DEBUG - [memory_event_storage.py] - memory_event_storage - 26 - Memory storage created with following configuration:
Max size: 100000
 Read records per time: 100"
""2020-12-01 13:06:58" - DEBUG - [tb_utility.py] - tb_utility - 96 - ModuleSpec(name='ModbusConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x757de030>, origin='/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/modbus/modbus_converter.py')"
""2020-12-01 13:06:58" - DEBUG - [tb_utility.py] - tb_utility - 103 - <module 'ModbusConnector' from '/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/modbus/modbus_converter.py'>"
""2020-12-01 13:06:58" - DEBUG - [tb_utility.py] - tb_utility - 96 - ModuleSpec(name='ModbusConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x757358f0>, origin='/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py')"
""2020-12-01 13:06:58" - DEBUG - [tb_utility.py] - tb_utility - 103 - <module 'ModbusConnector' from '/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py'>"
""2020-12-01 13:06:58" - DEBUG - [tb_utility.py] - tb_utility - 107 - Import ModbusConnector from /usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/modbus."
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 686 - Saved connected devices."
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 147 - Connected device Serial Machine"
""2020-12-01 13:06:58" - INFO - [modbus_connector.py] - modbus_connector - 70 - Starting Modbus connector"
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 663 - Loaded devices:
 {'Serial Machine': 'Modbus Default Server'}"
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 361 - Send data Thread has been started successfully."
""2020-12-01 13:06:58" - INFO - [tb_gateway_service.py] - tb_gateway_service - 135 - Gateway started."
""2020-12-01 13:06:58" - DEBUG - [modbus_connector.py] - modbus_connector - 237 - Modbus trying connect to Serial Machine"
""2020-12-01 13:06:58" - DEBUG - [modbus_connector.py] - modbus_connector - 239 - Modbus connected."
""2020-12-01 13:06:58" - DEBUG - [tb_client.py] - tb_client - 83 - TB client <paho.mqtt.client.Client object at 0x757357d0> connected to ThingsBoard"
""2020-12-01 13:06:58" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 147 - Connected device Serial Machine"
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2020-12-01 13:06:58" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2020-12-01 13:06:58" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 301 - Subscribed to * with id 2"
""2020-12-01 13:06:58" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 2"
""2020-12-01 13:06:59" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 233 - Received data: {}"
""2020-12-01 13:07:09" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 125 - Disconnected client: <paho.mqtt.client.Client object at 0x757357d0>, user data: None, result code: 1"
""2020-12-01 13:07:19" - DEBUG - [tb_client.py] - tb_client - 83 - TB client <paho.mqtt.client.Client object at 0x757357d0> connected to ThingsBoard"
""2020-12-01 13:07:19" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-12-01 13:07:19" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 147 - Connected device Serial Machine"
[... cut ...]
""2020-12-01 13:09:23" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2020-12-01 13:09:23" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2020-12-01 13:09:23" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 301 - Subscribed to * with id 9"
""2020-12-01 13:09:23" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 9"

Jaasdsa, 2020-12-24:
[image]
【wordOrder】大小写

zbeacon, 2021-01-05:
Hi @io-r-ahmed ,

type long in your config equals to 32uint, but at the same time 32uint are take 2 registers, so the gateway cannot parse received data. Please try to change type parameter in your configuration file to 16uint and probably you will be able to read the value.

malinglb, 2021-02-25:
Hi Gents, I am new in to Thingsboard(TB) and trying to connect TB CE into gateway and MQTT. Could you please help me crack the issue below:

cat logs/tb-connection.log
[image]

My TB-Gateway Restart Logs 

root@59d6703b21db:/thingsboard_gateway/config# python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("./tb_gateway.yaml")'
[STREAM ONLY] 2021-02-25 11:46:54,258 - ERROR - [mqtt_connector.py] - mqtt_connector - 130 - 'attributeRequests' section missing from configuration
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 3452, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 1181, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 2936, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 3216, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.5.1-py3.7.egg/paho/mqtt/client.py", line 3444, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 382, in _on_message
    found_device_name = TBUtility.get_value(handler["deviceNameJsonExpression"], content)
  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.5.4-py3.7.egg/thingsboard_gateway/tb_utility/tb_utility.py", line 124, in get_value
    body = loads(body)
  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.2-py3.7-linux-x86_64.egg/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.2-py3.7-linux-x86_64.egg/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.2-py3.7-linux-x86_64.egg/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)

malinglb, 2021-02-26:
@io-r-ahmed @zbeacon  Note that 'NoneType' object has no attribute 'upper' is one of the 2 parsing issues I experienced after tackling JSONDecoderError from MQTT.JSON config file.

malinglb, 2021-02-28:
Situation: I have setup Thingsboard community edition (CE) in my local development environment and I want to connect Thingsboard CE to an external real-time messaging system (Pub/Sub or Kafka, etc.) 
Task: As dev, I want to subscribe telemetry messages down from a device that generates MQTT messages to TB server, and later, use IoT rules engine to send actuation/query messages from the TB server to the devices.

Action: I have Thingsboard CE server running both locally and Demo instance. I used installation guide to configure a Thingsboard gateway that allow communication through MQTT connector and I have an MQTT client that simulate a device publishing topics and telemetry. The gateway provision is based on two files respectively tb-gateway.yaml (Contain Thingsboard CE server endpoint & redirect to mqtt.json) and mqtt.json (Contain all topic structure & endpoint to the Manager/MQTT)
https://thingsboard.io/docs/iot-gateway/getting-started/

Results:  On Thingsboard server screen, gateway device latest telemetry shows mqttbrokerconnectoreventproduced and mqttbrokerconnectoreventsent along eventproduced are 0 and logs/tb-connection.log shows subscription success to the targeted topic published by MQTT client. On gateway console, during gateway restart there’s a simple JSON decoder error. And “‘NoneType’ object has no attribute” (Being recreated). However, I am able to post telemetry from the gateway using curl API client as in hard coded and it appears On Thingsboard server screen as attribute and value of the gateway device latest telemetry.

malinglb, 2021-03-02:
kindly note more logs for this ticket.

cat /logs/connector.log

                                                                                                                                                                                                              ""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted mapping handlers: 3"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected mapping handlers: 0"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted serverSideRpc handlers: 2"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected serverSideRpc handlers: 0"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted connectRequests handlers: 2"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected connectRequests handlers: 0"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted disconnectRequests handlers: 2"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected disconnectRequests handlers: 0"
""2021-03-01 11:07:07" - ERROR - [mqtt_connector.py] - mqtt_connector - 130 - 'attributeRequests' section missing from configuration"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted attributeUpdates handlers: 1"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected attributeUpdates handlers: 0"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 230 - MQTT Broker Connector connected to 10.0.1.156:1883 - successfully."
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 271 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 271 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 271 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscription message id = 1"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 2"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 3"
""2021-03-01 11:07:07" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscription message id = 4"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted mapping handlers: 3"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected mapping handlers: 0"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted serverSideRpc handlers: 2"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected serverSideRpc handlers: 0"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted connectRequests handlers: 2"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected connectRequests handlers: 0"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted disconnectRequests handlers: 2"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected disconnectRequests handlers: 0"
""2021-03-01 11:14:08" - ERROR - [mqtt_connector.py] - mqtt_connector - 130 - 'attributeRequests' section missing from configuration"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 155 - Number of accepted attributeUpdates handlers: 1"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 159 - Number of rejected attributeUpdates handlers: 0"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 230 - MQTT Broker Connector connected to 10.0.1.156:1883 - successfully."
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 271 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 271 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 271 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscription message id = 1"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 2"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 3"
""2021-03-01 11:14:08" - INFO - [mqtt_connector.py] - mqtt_connector - 319 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscription message id = 4"
root@containerid:/#

malinglb, 2021-03-02:
Also kindly note below directory data/file content where I save the payload for each subscription to MQTT as per my tb_gateway.yaml configure the file is empty. 

[image]

root@containerid:/thingsboard_gateway/config# ls -al data/
total 12
drwxr-xr-x 2 root root 4096 Mar  2 14:49 .
drwxrwxr-x 3 root root 4096 Mar  2 14:22 ..
-rw-r--r-- 1 root root    0 Mar  1 09:27 data_1614590841325.txt
-rw-r--r-- 1 root root    0 Mar  1 09:27 data_1614590873469.txt
-rw-r--r-- 1 root root    0 Mar  2 14:19 data_1614694753633.txt
-rw-r--r-- 1 root root    0 Mar  2 14:24 data_1614695073096.txt
-rw-r--r-- 1 root root    0 Mar  2 14:33 data_1614695620269.txt
-rw-r--r-- 1 root root    0 Mar  2 14:36 data_1614695799897.txt
-rw-r--r-- 1 root root    0 Mar  2 14:48 data_1614696484551.txt
-rw-r--r-- 1 root root    0 Mar  2 14:48 data_1614696517914.txt
-rw-r--r-- 1 root root    0 Mar  2 14:49 data_1614696571466.txt
-rw-r--r-- 1 root root   49 Feb 24 05:51 state_file.txt
root@containerid:/thingsboard_gateway/config#

zbeacon, 2021-03-03:
Hi @malinglb ,

Thank you for your interest in ThingsBoard IoT gateway, but your question doesn't relate to the general topic of this issue. It looks like you have an issue with your payload from external broker - the gateway cannot parse it as a json, so it cannot convert the incoming value. Please check it and make sure that incoming data from external broker is json formatted. If you have different payload, you should use custom converter to parse data.

malinglb, 2021-03-03:
@zbeacon thank you for the feedback. Yes, I'm getting payload from an external broker and I formatted the message as json. Will try to publish to the custom converter.

closed 2021-03-22 by zbeacon