thingsboard/thingsboard-gateway #433: [BUG] Complex ack decoding error: InvalidTag('invalid tag encoding')

ingempo, 2020-11-30:
**Describe the bug**
We are not sure what it is the problem exactly. It seems a problem in Data Link Level. The application fails and no receive data. If we restart the application. The data link is re-established, and we get again the data.

**Connector name (If bug in the some connector):**
BACnet connector [appservice.py]

**Error traceback (If available):**

""2020-11-28 15:19:35" - DEBUG - [bacnet_uplink_converter.py] - bacnet_uplink_converter - 40 - <BACnetUplinkConverter.BACnetUplinkConverter object at 0x7f5e09aee0d0> {'deviceName': 'BACnet Device AirLab', 'deviceType': 'default', 'attributes': [], 'telemetry': [{'Humidity': 43.20000076293945}]}",
""2020-11-28 15:19:35" - DEBUG - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - 140 - Received response: <bacpypes.apdu.ReadPropertyACK(12,184) instance at 0x7f5e0829e990>",
""2020-11-28 15:19:35" - DEBUG - [bacnet_uplink_converter.py] - bacnet_uplink_converter - 40 - <BACnetUplinkConverter.BACnetUplinkConverter object at 0x7f5e09aee150> {'deviceName': 'BACnet Device AirLab', 'deviceType': 'default', 'attributes': [], 'telemetry': [{'CO2': 400.0}]}",
""2020-11-28 15:20:05" - DEBUG - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - 140 - Received response: <bacpypes.apdu.ReadPropertyACK(12,185) instance at 0x7f5e082a31d0>",
""2020-11-28 15:20:05" - DEBUG - [bacnet_uplink_converter.py] - bacnet_uplink_converter - 40 - <BACnetUplinkConverter.BACnetUplinkConverter object at 0x7f5e09aee1d0> {'deviceName': 'BACnet Device AirLab', 'deviceType': 'default', 'attributes': [], 'telemetry': [{'Temperature': 18.700000762939453}]}",
""2020-11-28 15:20:05" - ERROR - [appservice.py] - appservice - 1532 - complex ack decoding error: InvalidTag('invalid tag encoding')",
Traceback (most recent call last):,
  File "/root/.local/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 175, in decode,
    self.tagData = pdu.get_data(self.tagLVT),
  File "/root/.local/lib/python3.7/site-packages/bacpypes/comm.py", line 131, in get_data,
    raise DecodingError("no more packet data"),
bacpypes.errors.DecodingError: no more packet data,
,
During handling of the above exception, another exception occurred:,
,
Traceback (most recent call last):,
  File "/root/.local/lib/python3.7/site-packages/bacpypes/appservice.py", line 1530, in confirmation,
    xpdu.decode(apdu),
  File "/root/.local/lib/python3.7/site-packages/bacpypes/apdu.py", line 708, in decode,
    self._tag_list.decode(apdu),
  File "/root/.local/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 444, in decode,
    self.tagList.append( Tag(pdu) ),
  File "/root/.local/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 68, in __init__,
    self.decode(args[0]),
  File "/root/.local/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 177, in decode,
    raise InvalidTag("invalid tag encoding"),
bacpypes.errors.InvalidTag: invalid tag encoding,

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.4
 - Python version 3.8

zbeacon, 2020-12-01:
Hi @ingempo ,

I'm glad to see you here again. Thank you for participating in project life. 

I will change connection logic and push fix as soon as possible.

ingempo, 2020-12-01:
Hey @zbeacon ! Thank you for you support. We are so excited to test the APP. We have several BACnet devices. Please if you update the project, make sure thar the docker container is updated too. We work with Docker on Linux. Thank you again !

zbeacon, 2020-12-04:
Hi @ingempo ,

Could you share your configuration, here or send it to my email ibarkov@thingsboard.io 
I need your BACnet connector configuration for investigation purposes, because I cannot repeat the issue.

ingempo, 2020-12-18:
Hi @zbeacon !

I send you the files. Hope its helps to fix the bug.

closed 2021-11-15 by zbeacon