thingsboard/thingsboard-gateway #434: [HELP] Request Connector Custom Converter Error

yeduy, 2020-12-02:
Hello TB lovers,

We are trying to get data about the status of the lights using Request Connector from a gateway that we use for a lighting project. When we send a query to Gateway via Postman, we get an answer like p=value (p=1, p=2, p=3, p=4). The value after Equals indicates the state of light for us. So we aim to assign the value to a key and send it to Thingsboard. 

[image]

Since we know the format of the response (p=value) is not json, we set the config file to use the custom converter. However, we get an error and are unable to move forward. We think the source of the problem is related to the format of the response, but we are not entirely sure. We would appreciate it if you could at least help us understand the source of the problem.

[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

`
{
  "host": "http://192.168.70.116",
  "SSLVerify": false,
  "security": {
      "type": "basic",
      "username": "user",
      "password": "user"
  },
  "mapping": [
    {
      "url": "GetDyNet.cgi?a=2",
      "httpMethod": "GET",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 5,
      "scanPeriod": 5,
      "converter": {
        "type": "custom",
        "deviceNameJsonExpression": "SD8500",
        "deviceTypeJsonExpression": "SD",
        "extension": "CustomRequestUplinkConverter",
        "extension-config": [
          {
            "key": "Flow",
            "type": "int",
            "fromByte": 2,
            "toByte": 3,
            "byteorder": "big",
            "signed": true,
            "multiplier": 1
          }
        ]
      }
    }
  ],
  "attributeUpdates": [
      {
        "httpMethod": "POST",
        "httpHeaders": {
          "CONTENT-TYPE": "application/json"
        },
        "timeout": 0.5,
        "tries": 3,
        "allowRedirects": true,
        "deviceNameFilter": "SD.*",
        "attributeFilter": "send_data",
        "requestUrlExpression": "sensor/${deviceName}/${attributeKey}",
        "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
      }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "httpMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    }
  ]
}

`

**Connector name (If you need help with some connector/converter):**
[Request Connector]

**Error traceback (If it was raised):**

""2020-12-02 10:55:31" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-12-02 10:55:31" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 10"
""2020-12-02 10:55:33" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2020-12-02 10:55:33" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 11"
""2020-12-02 10:55:35" - ERROR - [request_connector.py] - request_connector - 215 - Expecting value: line 1 column 1 (char 0)"
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.5.4-py3.6.egg/thingsboard_gateway/connectors/request/request_connector.py", line 197, in __send_request
    data_to_storage.append(response.json())
  File "/usr/lib/python3/dist-packages/requests/models.py", line 892, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [master branch]
 - Python version[3.6.9]

Thank you

zbeacon, 2020-12-02:
Hi @yeduy ,

You have configuration with header "Accept: application/json", but you pass bytes, so the gateway cannot convert response data to json.

yeduy, 2020-12-02:
Hi @zbeacon,

Thank you for your answer. We followed this guide: https://thingsboard.io/docs/iot-gateway/config/request/?RequestConverterTypeConfig=custom and in the example here it was "Accept: application / json".

We are not very knowledgeable about this. Could you help with what to write in the "Accept" section?

Thank you

zbeacon, 2020-12-02:
You may try to use `*/*` instead of `application/json`.

closed 2020-12-22 by yeduy