thingsboard/thingsboard-gateway #435: [BUG]The Gateway issues the RPC command

chenaiyuan, 2020-12-02:
When the RPC command is issued, it is found that the command is issued normally, but the parameters (params) at the time of issuing cannot be passed through.（The latest code）

It turned out to be a bit of a bug in the code,Location is:
thingsboard_gateway/connectors/request/request_connector.py  Line 186： "data": request["config"].get("content", {})  
The 'content' should be 'data'.

But after the correction or not, I wonder if some places have not been completely changed.

Looking forward to your reply.

zbeacon, 2020-12-04:
Hi @chenaiyuan ,

I replaced this parameter in gateway, you was right. Thank you for investigations. You may try the version from the master branch, it contains this fix.

chenaiyuan, 2020-12-04:
Hello, have you tested it? May I ask when the package of RPM can be produced? Because I always have a problem with source code packaging.

chenaiyuan, 2020-12-04:
> 嗨@chenaiyuan，
> 
> 我在网关中替换了此参数，您是对的。感谢您的调查。您可以从master分支尝试该版本，它包含此修复程序。

Hello, have you tested it? May I ask when the package of RPM can be produced? Because I always have a problem with source code packaging.

chenaiyuan, 2020-12-04:
> Hi @chenaiyuan ,
> 
> I replaced this parameter in gateway, you was right. Thank you for investigations. You may try the version from the master branch, it contains this fix.

Hello, I just looked at the place you changed, and I also changed it in the same way, but I have tested it. It seems that only this way was changed, but the call still failed. Could you please have a look and tell me what the problem is? Thank you very much.