thingsboard/thingsboard-gateway #438: [BUG] MQTT RPC Request through gateway

a-magri, 2020-12-08:
**Describe the bug**
Hi all, im pretty new to Thingsboard and Thingsboard IoT Gateway.
Im able to push telemetry and attributes through gateway in Thingsboard using Mosquitto as broker configured as MQTT Connector, but im struggling sending RPC commands from Thingsboard to the gateway and from gateway to device.
To send the RPC request, im using a "Send RPC" button in the Thingsboard dashboard from Control Widget with entity alias of gateway device.
The button sends a simple message "Hello World", the option "Is one way command" is checked and the method name is "helloWorld".
From Thingsboard, in the "Log audit" of the gateway device i can check that the message is sent successfully, but from the gateway the message is not sent to the broker through the topic (requestTopicExpression) specified in the connector (mqtt.json).

I tried subscribing with mosquitto_sub to several topics, included topics described in the MQTT Gateway API Reference in Thingsboard documentation with gateway access token as authentication, but nothing happens.
```
sensor/SN-001/response/helloWorld/+
sensor/SN-001/response/no-reply/+
sensor/SN-001/response/helloWorld/#
sensor/SN-001/response/no-reply/#
v1/gateway/rpc
```

**Connector name (If bug in the some connector):**
Im using an MQTT Connector with  "serverSideRPC" configured in this way:
```
"serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
```
I tried to set the "requestTopicExpression" to the following: "sensor/request" and i've added the following parameters too to try if something would have changed, but still nothing.
```
{
      "deviceNameFilter": "SN-001",
      "methodFilter": "helloWorld",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
```
**Error traceback (If available):**
When i send the RPC command, i noticed that checking the gateway service log and the status of the gateway with:
```
systemctl status thingsboard-gateway
```
these error comes out:
```
ERROR - [tb_gateway_service.py] - tb_gateway_service - 497 - 'NoneType' object has no attribute 'shell_commands'"
 Traceback (most recent call last):
 File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 484, in _rpc_request_handler
     elif module == 'gateway' or module in self.__remote_shell.shell_commands:
 AttributeError: 'NoneType' object has no attribute 'shell_commands'
```
Activating the remote shell in the gateway, i get a different error:
```
ERROR - [tb_gateway_service.py] - tb_gateway_service - 487 - Connector "hello" not found
```
**Versions (please complete the following information):**
The device i have used to install the gateway is a RaspberryPi4 which is installed Mosquitto broker too.
 - OS: Raspbian GNU/Linux 10 (buster)
 - Thingsboard IoT Gateway version: 2.5.5.2 
 - Python version: 3.7.3

zbeacon, 2020-12-31:
Hi @a-magri ,

Thank you for your interest in Thingsboard IoT gateway. It looks like you try to send RPC request to gateway device, but you should send it to device, it will be redirected to gateway automatically.

a-magri, 2020-12-31:
Thanks a lot for the reply it works really well now.

I've followed this old topic #125 in which you wrote to send the RPC request to the gateway.
I love ThingsBoard and ThingsBoard Gateway, but the documentation doesn't specify the RPC flow, and lacks of some contents.
Please improve the documentation adding those details becouse for unexperienced users like me, it's really important.

Thanks again for all the awesome work.

closed 2021-01-18 by zbeacon