thingsboard/thingsboard-gateway #443: AttributeError: 'str' object has no attribute 'registers' using Modbus Connector

abhinavrawat27, 2020-12-18:
Hi

I have setup iot gatway on raspberry pi. I want to use thingsboard modbus connector to connect to modbus tcp/ip server.

For this I have modified the modbus.json file, which is below:

    {
      "server": {
        "name": "Modbus Default Server",
        "type": "tcp",
        "host": "192.168.0.102",
        "port": 502,
        "timeout": 35,
        "method": "socket",
        "devices": [
          {
            "unitId": 1,
            "deviceName": "raspberry",
            "timeseriesPollPeriod": 5000,
            "sendDataOnlyOnChange": false,
            "timeseries": [
              {
                "byteOrder": "BIG",
                "tag": "humidity",
                "type": "16int",
                "functionCode": 4,
                "registerCount": 1,
                "address": 1019
              },
              {
                "byteOrder": "BIG",
                "tag": "temperature",
                "type": "16int",
                "functionCode": 4,
                "registerCount": 1,
                "address": 1020
              }
            ]
          }
        ]
      }
    }

But I kept getting this error:

    Dec 18 18:42:12 raspberrypi python3[2695]: ""2020-12-18 18:42:12" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 89 - 'str' object has no attri
    Dec 18 18:42:12 raspberrypi python3[2695]: Traceback (most recent call last):
    Dec 18 18:42:12 raspberrypi python3[2695]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 68, in convert
    Dec 18 18:42:12 raspberrypi python3[2695]:     registers = response.registers
    Dec 18 18:42:12 raspberrypi python3[2695]: AttributeError: 'str' object has no attribute 'registers'
    Dec 18 18:42:12 raspberrypi python3[2695]: ""2020-12-18 18:42:12" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 89 - 'str' object has no attri
    Dec 18 18:42:12 raspberrypi python3[2695]: Traceback (most recent call last):
    Dec 18 18:42:12 raspberrypi python3[2695]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 68, in convert
    Dec 18 18:42:12 raspberrypi python3[2695]:     registers = response.registers
    Dec 18 18:42:12 raspberrypi python3[2695]: AttributeError: 'str' object has no attribute 'registers'

So to test the connection with my modbus tcp/ip server, I wrote the test python code:

     from pymodbus.client.sync import ModbusTcpClient

	client = ModbusTcpClient('192.168.0.102')
	connection = client.connect()

	data1 = client.read_holding_registers(0x03FB, 1, unit=1)   # 1019
	data1_value = data1.registers

	data2 = client.read_holding_registers(0x03FC, 1, unit=1)  # 1020
	data2_value = data2.registers

	print(data1_value, data2_value)

and I got the correct response  `[12] [13]`

Now I am not sure why the thingsboard modbus connector is throwing the error. Can anyone please help. Thanks

zbeacon, 2021-01-05:
Hi @abhinavrawat27 ,

Thank you for your interest in ThingsBoard IoT gateway, it was an error in exception handling. Please try the version from master branch and run your gateway again, it will show you the real error.

zbeacon, 2021-01-05:
Also I have to say that your configuration looks correct.

closed 2021-03-01 by zbeacon