thingsboard/thingsboard-gateway #456: [HELP] Thingsboard gateway modbus json configuration problem

leeweizhang1006, 2020-12-29:
**Describe the issue**
Basically I want to use thingsboard and thingsboard gateway to control my relay module and show the result in thingsboard.
I have follow this link to download thingsboard and thingsboard gateway on my raspberry pi running ubuntu lts 20.4 
https://thingsboard.io/docs/iot-gateway/install/deb-installation/
https://thingsboard.io/docs/user-guide/install/rpi/
After that I go to edit my tb_gateway.yaml and modbus.json file which I will attach on below

After I edit the both file I get some error message I will also include in below, other then that the attribute and rpc that I configure in modbus.json also didn't show in device details.

**Configuration**
**tb_gateway.yaml*
thingsboard:
  host: 192.168.1.xxx
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: xxxxxxxxxxx
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 10000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
  
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus.json

**modbus.json**
{
  "server": {
    "type": "tcp",
    "host": 192.168.1.116",
    "port": 1234,
    "timeout": 35,
    "method": "socket",
    "byteOrder": "BIG",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "RelayDemo"
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "tag": "Relay",
            "type": "boolean",
            "functionCode": 1,
            "objectCount": 1,
            "address": 0,
          }
      ],
        "rpc": [
          {
            "tag": "setValue",
            "type": "boolean",
            "functionCode": 15,
            "objectCount": 1,
            "address": 0,
          }
      ],

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
When I was running systemctl status thingsboard-gateway.service, I get the error below
Dec 29 02:39:32 ubuntu python3[17234]:      self.client.framer.processIncomingPacket(response,
Dec 29 02:39:32 ubuntu python3[17234]:
Dec 29 02:39:32 ubuntu python3[17234]:

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

wudehua2016, 2021-02-14:
yes, I got this problem, too. Maybe it is a bug?

zbeacon, 2021-03-02:
Hello, everybody. 

Thank you for your interest in ThingsBoard IoT gateway.
It looks like an error with device - it sent response with 5 bytes, but the Modbus library expects at least 8 bytes. 
Also it looks like you have an issue with your configuration, because we don't have "boolean" as a type. You can read more [here](https://thingsboard.io/docs/iot-gateway/config/modbus/#data-types). I hope it will be helpful. Also I apologize for long-term answer, we have a lot of different tasks in the general ThingsBoard. 
If you will still get an error - please open the new issue.

closed 2021-03-02 by zbeacon