thingsboard/thingsboard-gateway #457: [HELP] Error 104, Connection reset by peer

yvo-pfenninger, 2021-01-05:
Not possible to write values to the gateway. I once worked with 2.5.2.

ErrorLOG:

""2020-12-23 10:43:17" - DEBUG - [bytes_modbus_downlink_converter.py] - bytes_modbus_downlink_converter - 95 - [11547]"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 301 - With result Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 371 - ModbusIOException(ConnectionResetError(104, 'Connection reset by peer'), 6)"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 237 - Modbus trying connect to TCP_Slave005
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 239 - Modbus connected."
""2020-12-23 10:43:17" - DEBUG - [bytes_modbus_downlink_converter.py] - bytes_modbus_downlink_converter - 95 - [0]"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 301 - With result Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 371 - ModbusIOException(ConnectionResetError(104, 'Connection reset by peer'), 6)"
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 237 - Modbus trying connect to  TCP_Slave005
""2020-12-23 10:43:17" - DEBUG - [modbus_connector.py] - modbus_connector - 239 - Modbus connected."

Versions (please complete the following information):

Debian GNU/Linux 10 (buster)
Kernel: 5.4.24
docker/stable 1.5-2 all
Thingsboard IoT Gateway version 2.5.5.2
python3/stable 3.7.3-1 armhf

current modbus.json config file

{
  "server": {
    "name": "Modbus Default Server",
    "type": "tcp",
    "host": "192.168.11.152",
    "port": 502,
    "timeout": 35,
    "method": "socket",
    "byteOrder": "BIG",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "TCP_Slave005",
        "deviceType": "Type005",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "cable_state",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1004,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "firmware_code",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1013,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "charging_state",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1000,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "serial_number",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1014,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "feature_code",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1016,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "max_supported_current",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1110,
            "deviceName": "TCP_Slave005"
          }
        ],
        "timeseries": [
          {
            "byteOrder": "BIG",
            "tag": "tt_forward_active_energy",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1036,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "total_active_power",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1020,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "tt_forward_active_energy_thisSession",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1502,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "charging_state",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1000,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "cable_state",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1004,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "error_code",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1006,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "l1_current",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1008,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "l2_current",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1010,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "l3_current",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1012,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "RFID_card",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1500,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "l1_voltage",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1040,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "l2_voltage",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1042,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "l3_voltage",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1044,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "power_factor",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1046,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "max_charging_current",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 1100,
            "deviceName": "TCP_Slave005"
          }
        ],
        "attributeUpdates": [
          {
            "byteOrder": "BIG",
            "tag": "set_charging_current",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 5004,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "set_energy",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 5010,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "unlock_plug",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 5012,
            "deviceName": "TCP_Slave005"
          },
          {
            "byteOrder": "BIG",
            "tag": "enable_user",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 5014,
            "deviceName": "TCP_Slave005"
          }
        ]
      }
    ]
  }
}

yvo-pfenninger, 2021-11-02:
@samson0v Thank you for your help! I am testing it on the newest docker. I will keep you updated.

samson0v, 2021-11-23:
Hi @yvo-pfenninger, did you test the Gateway?